xtacocorex/CHIP_IO #4: CHIP_IO.SPI ?

eparadis, 2016-05-10:
Is there any possibility of supporting something like Adafruit's `Adafruit_GPIO.SPI` module?
I'm looking to try their [user space ILI9341 module](https://github.com/adafruit/Adafruit_Python_ILI9341) to control a [Seeedstudio TFT Touch Shield v2.0](http://www.seeedstudio.com/wiki/2.8''_TFT_Touch_Shield_v2.0).

I'm new to the CHIP, and [it seems there are SPI ports available](http://www.chip-community.org/index.php/SPI_support) but perhaps they require a custom kernel.

xtacocorex, 2016-05-10:
While CHIP_IO is able to be run by itself, it is used by my port of Adafruit_GPIO.  The Adafruit code for SPI is not computer specific, which is why I decided not to wrap spidev in more code.

I haven't tested SPI on my CHIP, but I do have a custom kernel that displays the spidev device in /dev.

I'm very certain that my Adafruit_GPIO port would support SPI on the CHIP without mods.

Looking at the ILI9341 code, it requires GPIO control also, so you would need both CHIP_IO and my Adafruit_GPIO port.

xtacocorex, 2016-11-17:
Going to close this as spidev meets the need for SPI devices.

closed 2016-11-17 by xtacocorex