xtacocorex/CHIP_IO #5: new_xio: Channel in use when running as non-root

maweki, 2016-07-03:
If you're not root (or probably not in the group that is allowed), and try to set up a pin, the following happens:

  >>> import CHIP_IO.GPIO as GPIO
  >>> GPIO.setup("GPIO0", GPIO.OUT)
  Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
  ValueError: Channel in use (already exported)

Even though I am just not allowed to and the channel is actually not exported.

xtacocorex, 2016-07-03:
GPIO0 is not allowed as a valid input.  All scripts accessing the GPIO have to be run as root.  I2C is a different story as the chip user is in the i2c group and doesn't need to access the bus with super user permissions.  GPIO is different as there is no group associated with the interface.

fordsfords, 2016-07-03:
@maweki , is your suggestion that we have a more user-friendly error message?  Or that we arrange it so that access is controlled by a group ID, rather than root?

maweki, 2016-07-04:
@fordsfords the thing I stumbled over was the error message. I thought "Channel in use" is a misleading error message when its decidedly not in use but I am just not allowed to set the channel up.

I have yet to try other interactions. I'd prefer if I could put my user in a group and allow controlling the pins from there but that's a different issue.

xtacocorex, 2016-07-04:
Yeah, I see where you're coming from @maweki.  I was re-creating your code above as I was working through the issues.  The code isn't throwing the Invalid GPIO exception that I wanted and it's due to how the GPIO pin is set when the code starts setting everything up.

fordsfords, 2016-07-04:
Ok, I've checked in a bunch of improvements to error handling.  I've also added tests to pytest.py

I think doing work with group permission is beyond the scope of this project.

xtacocorex, 2016-07-06:
Error checking looks good.  Definitely agree that the permissions stuff is beyond our control.

fordsfords, 2016-07-09:
OK, for completeness, I tried the sample that was given above:

>>> import CHIP_IO.GPIO as GPIO
>>> GPIO.setup("GPIO0", GPIO.OUT)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Invalid channel GPIO0. ()

This is a better error message.

closed 2016-07-09 by fordsfords