xtacocorex/CHIP_IO #6: new_xio: Error when Pin-Name not correct

maweki, 2016-07-03:
If you set-up a pin via string name (I think there should be constants available as well, but that's just me), one gets the following error:

```
>>> import CHIP_IO.GPIO as GPIO
>>> GPIO.setup("GPIO0", GPIO.OUT)
ASSRT failed at source/event_gpio.c:248 (s == strlen(direction))
```

While "XIO-P0" works fine, "GPIO0" throws this c-error.

xtacocorex, 2016-07-03:
GPIO0 is not allowed, so the error is legit.  The allowable names are what are defined by NTC on their documentation or the header and pin number.  I'll update the README to include the allowable names for setup.

maweki, 2016-07-03:
Of course it is a legit error. But if I enter the wrong string, there should maybe be an exception with a good error message instead of a C-Error that crashes the interpreter, no?

xtacocorex, 2016-07-03:
Touche'

Totally see what you're getting at now.  Will work to update that.

fordsfords, 2016-07-03:
> there should maybe be an exception with a good error message

That's on me.  I added a lot of error checking, but I should only have used abort for "internal errors" (i.e. internal inconsistancies) rather than for user errors.

xtacocorex, 2016-07-06:
This was fixed in #7.  Closing this.

@maweki just know that I followed the GPIO numbers from the PocketCHIP header, so it's GPIO1-6 for the alternate names.

closed 2016-07-06 by xtacocorex

maweki, 2016-07-06:
So `GPIO.setup("foo", GPIO.OUT)` doesn't crash the interpreter?

fordsfords, 2016-07-06:
Correct.  If you use the new_xio branch, it should return a python exception with a user-friendly message.  But by all means give it a test in case I forgot something.