xtacocorex/CHIP_IO #7: Add pin aliases

maweki, 2016-07-04:
As discussed in #6 , "GPIO0" is not an allowed pin name.

> The allowable names are what are defined by NTC on their documentation or the header and pin number.

The PocketCHIP does have a breakout of some pins at the top and there are indeed pins labeled "GPIO1" to "GPIO6". So it would be cool if these alternate names would be allowed.

fordsfords, 2016-07-05:
I'll try to get this in during the week, otherwise the weekend.

xtacocorex, 2016-07-06:
Started implementing this, since it shouldn't be too difficult for my weak C skills. :)

xtacocorex, 2016-07-06:
In the process, I've changed around gptest.py a bit, to allow us to test the alternate names, so anything with XIO-P0 moved to XIO-P2/GPIO1.

xtacocorex, 2016-07-06:
@maweki Just pushed up code (new_xio branch) to handle the alternate names if you wanted to give it a shot.  Going to go ahead and close this as I have been able to successfully command GPIO1 and GPIO6:

`Python 2.7.9 (default, Mar  1 2015, 13:48:22) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import CHIP_IO.GPIO as GPIO
>>> GPIO.setup("GPIO1",GPIO.OUT)
>>> GPIO.setup("GPIO6",GPIO.IN)
>>> GPIO.output("GPIO1",GPIO.HIGH)
>>> GPIO.input("GPIO6")
1
>>> GPIO.output("GPIO1",GPIO.LOW)
>>> GPIO.input("GPIO6")
0
>>> `

(disregard the formatting, it's crappy)

closed 2016-07-06 by xtacocorex