xtacocorex/CHIP_IO #8: py.test fail on softpwm: segmentation fault

nyboer, 2016-07-05:
trying this stuff out...I tried the py.test, with these results:
```
test/test_gpio_input.py FF
test/test_gpio_output.py FFFF..
test/test_gpio_setup.py FFFFFFF..F
test/test_pwm_setup.py FFFF...........F.........
test/test_softpwm_setup.py F...Segmentation fault
```

fordsfords, 2016-07-05:
Thanks, will look at it.  I think I know what might be wrong.

fordsfords, 2016-07-05:
Forgive my ignorance (I'm a Python newbie), but how exactly do you run these tests?  I.e. what directory do you "cd" to, and what command do you enter?  Does it just know to run the tests under the "test" directory?  Do all those "F" letters represent failures?

nyboer, 2016-07-05:
I was just following instructions at the bottom of the README. Install python test,  `cd test` then run `py.test`. Seemed worth noting as an issue that the test failed :)

xtacocorex, 2016-07-06:
py.test needs to be run with sudo, I've updated the README to denote that.

@fordsfords, yes, the F means failure.  This is leftover code from the BBB that I've updated and added to.  The scripts don't prompt you for what connections are required, so you have to look at the code to know what is required to have hooked up.

fordsfords, 2016-07-09:
OK, I fixed the softpwm bug.  So this should be fixed.

The stopping of a PWM channel did not synchronize with the thread.  It set the "stop_flag", but didn't wait for the thread to stop before deallocating the structures.  This allowed the thread to wake up and attempt to access those structures, leading to much evilness.

In the process, I found an other bug where I didn't account for a newline, and therefore didn't recognize a line read from the "direction" file.

I also tightened up a bit more error checking, and cleaned up the code a bit.  Note that I use "pymode" in vim, which enforces various coding standards, including things like where you put two blank lines and where you need spaces.

HOWEVER, I did *not* do anything with the hardware PWM, and I also did not do anything with the edge trigger stuff.

closed 2016-07-09 by fordsfords