xtacocorex/CHIP_IO #14: SOFTPWM does not do anything

ervion-n, 2016-07-24:
I have CHIP v1.0 and firmware official build headless 4.4. 

Simply toggling gpio works using this library, but soft pwm does nothing. I have measured both with multimeter and ostilloscope and detected no change in output. I did not get any error messages either.

`import time
import CHIP_IO.SOFTPWM as PWM
PWM.start("XIO-P7", 50)
time.sleep(5)
PWM.stop("XIO-P7")
PWM.cleanup()
`

ervion-n, 2016-07-24:
However, if I try to change something, i get error 
```
Traceback (most recent call last):
  File "test.py", line 4, in <module>
    PWM.set_frequency("XIO-P7", 10)
RuntimeError: You must start() the PWM channel first
```
using following code:
```
import time
import CHIP_IO.SOFTPWM as PWM
PWM.start("XIO-P7", 50)
PWM.set_frequency("XIO-P7", 10)
time.sleep(5)
PWM.stop("XIO-P7")
PWM.cleanup()
```

aninternetof, 2016-07-25:
@ervion-n, I don't see anything wrong with your code. I'll dig in and try to see what the problem is tomorrow morning.

xtacocorex, 2016-07-25:
Agreed that the code looks good.  There is definitely something goofy going on as I don't get any data in my test code which I just committed.  It essentially creates a thread for listening to a PWM signal from a source.  Source is XIO-P7, Receiver is CSID0.  Just need a test lead hooked between the two pins.

aninternetof, 2016-07-25:
running it line by line, I am seeing it go into PWM.start("XIO-P7", 50) and never return. Now I'm trying to add some debug statements to see why it's getting hung up.

xtacocorex, 2016-07-25:
I didn't test this part of the library after @fordsfords did his mods. :( Maybe the errors were what made it work. :)  [ I had a FORTRAN program in college that would only work if I accessed an array out of bounds, otherwise I wouldn't get the right answer, it was so messed up and confused my friends and I ]

aninternetof, 2016-07-25:
Ok, looks like it's spinning [here in softpwm_set_duty_cycle](https://github.com/xtacocorex/CHIP_IO/blob/master/source/c_softpwm.c#L139), waiting for mutex lock. Now to figure out why...

aninternetof, 2016-07-25:
(obligatory "I swear I saw this working before I committed it")

aninternetof, 2016-07-25:
Ah, I think I see it. I don't unlock the mutex until the very end of softpwm_start(). Meanwhile, in the middle of softwpwm_start(), softpwm_set_duty_cycle() wants to acquire the lock

aninternetof, 2016-07-25:
Ok, it was two things:

- The mutex locking oversight described above
- softpwm_set_frequency() was returning 1 on success instead of 0, so the ASSRT was catching it

I'm about to push up my hotfix branch, but git is complaining "Could not resolve host: github.com" when I push. Not sure what's up with that. Hope to have it resolved shortly.

aninternetof, 2016-07-25:
Ok, there we go! @xtacocorex  or @ervion-n, would somebody mind pulling down the hotfix/softpwm_stall branch and confirm that it makes PWM work for you guys?

xtacocorex, 2016-07-25:
Will do when I get home from work, it'll probably be sometime tonight.

fordsfords, 2016-07-25:
> Ah, I think I see it. I don't unlock the mutex until the very end of softpwm_start(). Meanwhile, in the middle of softwpwm_start(), softpwm_set_duty_cycle() wants to acquire the lock

Oh no!  Was that my screw up?  I did change the mutexing a bit, and I thought I tested it pretty well!

xtacocorex, 2016-07-26:
No worries @fordsfords, there was a lot of stuff going on.  I just verified I get 0's and 1's out in the spwmtest.py script, so I'm going to go ahead and close and merge this to master.  I updated to v0.1.2.  Forgot the proper commit message to reference this issue. :(

closed 2016-07-26 by xtacocorex