xtacocorex/CHIP_IO #16: pwm problem

yorklai, 2016-07-26:
i use 4.4 or 4.3 firmware:
I follow your sample will found error:

>>> import CHIP_IO.SOFTPWM as SPWM
>>> SPWM.start("XIO-P7", 50)
>>> SPWM.set_duty_cycle("XIO-P7", 25.5)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
**RuntimeError: You must start() the PWM channel first**
>>>

ervion-n, 2016-07-26:
this is just recently fixed problem: https://github.com/xtacocorex/CHIP_IO/issues/14

xtacocorex, 2016-07-26:
Well, I didn't verify in my spwmtest.py script of changing the duty cycle that way, so there could still be an issue.  Will try to look this over tonight.

aninternetof, 2016-07-26:
Let me know what you find if you check it out tonight, @xtacocorex. I didn't retest that case either, but it can look at it tomorrow morning if it is indeed a legit bug.

xtacocorex, 2016-07-27:
@yorklai, what version of CHIP_IO are you using?

What happens if you do the following from the Python shell:
import CHIP_IO.GPIO as GPIO
print GPIO.VERSION

I'm running 0.2.1 and this isn't an issue. I believe @aninternetof fixed it yesterday for Issue#14.

yorklai, 2016-07-27:
I try Pocketchip (4.3 firmware):

Python 2.7.9 (default, Mar  1 2015, 13:48:22)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import CHIP_IO.GPIO as GPIO
>>> print GPIO.VERSION
0.2.1
>>>
>>>

try SOFTPWM sample code:

Python 2.7.9 (default, Mar  1 2015, 13:48:22)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import CHIP_IO.SOFTPWM as SPWM
>>> SPWM.start("XIO-P7", 50)
>>> SPWM.set_duty_cycle("XIO-P7", 25.5)Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: You must start() the PWM channel first
>>>

yorklai, 2016-07-27:
The C.H.I.P 4.4 firmware (any) have bug:

Pocketchip(4.3 firmware):

chip@chip:/sys/class/pwm/pwmchip0$ lsdevice  export  npwm  power
subsystem  uevent  unexport
chip@chip:/sys/class/pwm/pwmchip0$

C.H.I.P (4.4 Gui no limit firmware):

chip@chip:~$ cd /sys
chip@chip:/sys$ cd class
chip@chip:/sys/class$ cd pwm
chip@chip:/sys/class/pwm$ ls
chip@chip:/sys/class/pwm$ ls -l
total 0
chip@chip:/sys/class/pwm$

2016-07-27 19:54 GMT+08:00 York Lai <york.star@gmail.com>:

aninternetof, 2016-07-27:
@yorklai, regarding the "RuntimeError: You must start() the PWM channel first", I think I've figured out what is going on.

If you first tried with the old version of CHIP_IO (v0.1.1), then you hit a bug and the code exited. When it did this, it left the GPIO exported, because SPWM.cleanup() was never called.

So now when you try to run with the fixed code, the command `SPWM.start("XIO-P7", 50)` seems like it is running successfully, but it is failing to add XIO-P7 to the list of running PWMs because the gpio has already been exported.  Then when SPWM looks for it on the list of running PWMS, it can't find it and throws the RuntimeError.

I'll work on a commit that makes SPWM.start() throw an error instead of failing silently if the gpio has already been exported. To fix your problem you'll need to:
- `sudo echo 415 > /sys/class/gpio/unexport`
- Try your code again. Make sure to add SPWM.cleanup() to the end of your test script.

fordsfords, 2016-07-27:
I can answer that one.  It adds text to a global string buffer (error_msg_buff).  It is a way that a descriptive error message can be passed back up to the python layer.  So some utility function several layers down can detect a problem and add some text to that buffer.  It returns an error, and its caller might add some more text to give it context (i.e. what it was trying to do when it call the lower-level function which detected the error), and that caller returns an error.  And so on, until it makes it to the C code which interfaces with python.  At that point, a final error message string is formatted and passed back to Python, using get_error_msg().  For example, in py_gpio.c the function "static PyObject *py_setup_channel(..." contains the code:
`
   if (gpio_export(gpio) < 0) {
      char err[2000];
      snprintf(err, sizeof(err), "Error setting up channel %s, maybe already exported? (%s)", channel, get_error_msg());
      PyErr_SetString(PyExc_RuntimeError, err);
      return NULL;
   }
`
The built-up error message is included in parentheses.

yorklai, 2016-07-28:
Dear Sir,

     I boot pocket (4.3 firmware) then try:

chip@chip:/etc/NetworkManager/system-connections$ sudo echo 415 >
/sys/class/gpio/unexport
-bash: /sys/class/gpio/unexport: Permission denied
chip@chip:/etc/NetworkManager/system-connections$
chip@chip:/etc/NetworkManager/system-connections$ sudo su
root@chip:/etc/NetworkManager/system-connections# sudo echo 415 >
/sys/class/gpio/unexportecho: write error: Invalid argument

xtacocorex, 2016-07-28:
When you did the sudo su, you got root access.  sudo doesn't work when you're root.

Have you tried rebooting since you pulled the latest CHIP_IO code?
What happens if you do run spwmtest.py from within the test directory?  You need a test wire hooked between XIO-P7 and CSID0 for that to run.

If that script runs correctly, you should be good to go.

aninternetof, 2016-07-28:
@fordsfords, thanks! I need to add pretty much exactly that same get_err_msg() code to my py_softpwm.c file. That would have made @yorklai's problem much more easily identifiable. I'll try to commit that sometime today.

aninternetof, 2016-07-28:
Ok, I think I've got it. Would sometime mind looking over [my commit in hotfix/SPWM_Issue_16](https://github.com/xtacocorex/CHIP_IO/commit/1e3e801e60e5ad7477292d40a8f1b2ac763d44eb)?

aninternetof, 2016-07-28:
For this kind of situation, have we considered having a cleanup_all() function to unexport _all_ GPIO, instead of just the ones registered in CHIP_IO's currently exported list?
It would be a somewhat powerful/dangerous function, but during development it's not uncommon to exit your script before reaching the cleanup(), and then you're left with stranded GPIO that you have to manually unexport before trying your script again.

xtacocorex, 2016-07-28:
Will try to test your hotfix when I can, might not be able to get to it tonight though.

fordsfords, 2016-07-28:
Can I reply to this email?

I just wrote a simple shell script to close everything. I don't think a close all in the GPIO library is a good idea. But I won't object if I am out-voted. :-)

fordsfords, 2016-07-29:
Oh, I guess I already added that script to the repo.  Named "unexport_all.sh", it is in the parent directory.

aninternetof, 2016-07-29:
A script like that works for me! It's probably a good middle-ground.

xtacocorex, 2016-07-30:
Just tested with by exporting a GPIO and then trying to start SPWM on it.  Error messages are displayed.  I updated the spwmtest.py to check for this issue.  Will commit spwmtest.py, update the version/changelog stuff, then merge to master.

closed 2016-07-30 by xtacocorex

yorklai, 2016-08-01:
Very Thanks! I try last version is no any error message!

chip@chip:~$ sudo python
Python 2.7.9 (default, Mar  1 2015, 13:48:22) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import CHIP_IO.SOFTPWM as SPWM
>>> print SPWM.VERSION
0.2.2
>>> SPWM.start("XIO-P7", 50)
>>> SPWM.set_duty_cycle("XIO-P7", 25.5)
>>> SPWM.set_frequency("XIO-P7", 10)
>>> SPWM.stop("XIO-P7")
>>> SPWM.cleanup()
>>> SPWM.start("XIO-P7", 50, 2000, 1)
>>>