xtacocorex/CHIP_IO #17: HW PWM Issues

xtacocorex, 2016-07-27:
I just confirmed issues with the HW PWM code. 

@fordsfords, looks like your imporovements might have broken it. :(

Will commit my test code for the HW PWM.  Need to be running 4.4 with a custom kernel enable CONFIG_OF_CONFIGFS.

fordsfords, 2016-07-29:
Crud, sorry 'bout that.  Teaches me to make changes to code that I can't really test.

xtacocorex, 2016-07-30:
This has been fun to try and troubleshoot.  The whole PWM sysfs stuff is wonky.
I also seem to have trouble lining up my debug asterisks.
 
    LOADING PWM OVERLAY
    PWM START
     ** IN pwm_start: pwm_initialized = 0
    ** EXPORTING PWM0 **
     ** IN initialize_pwm: s = 1, len = 1
     ** IN pwm_start: pwm_initialized = 1
    ** IN pwm_start: pwm_base_path: /sys/class/pwm/pwmchip0/pwm0
    ** IN pwm_start: enable_path:   /sys/class/pwm/pwmchip0/pwm0/enable
    ** IN pwm_start: period_path:   /sys/class/pwm/pwmchip0/pwm0/period
    ** IN pwm_start: duty_path:     /sys/class/pwm/pwmchip0/pwm0/duty_cycle
    ** IN pwm_start: polarity_path: /sys/class/pwm/pwmchip0/pwm0/polarity
     ** IN pwm_start: IF WE MAKE IT HERE, THE FILES WERE SUCCESSFULLY OPEN **
     ** IN pwm_start: CALLING THE SET FUNCTIONS **
     ** IN pwm_set_frequency: pwm_initialized = 1
     ** pwm_set_frequency: buffer: 666666
     ** IN pwm_set_frequency: s = -1, len = 6
     ** pwm_set_frequency rtnval = 1
     ** IN pwm_set_polarity: pwm_initialized = 1
     ** pwm_set_poliarity: buffer: inverted
     ** IN pwm_set_polarity: s = -1, len = 8
     ** pwm_set_polarity rtnval = 0
     ** pwm_set_enable rtnval = 0
     ** IN pwm_set_duty_cycle: pwm_initialized = 1
     ** pwm_set_duty_cycle: buffer: 333333
     ** IN pwm_set_duty_cycle: s = -1, len = 6
     ** pwm_set_duty_cycle rtnval = 0
    PRINTING PWM SYSFS DATA
    PWM0 ENABLE:    0
    
    PWM0 PERIOD:    0
    
    PWM0 DUTY CYCLE:        0
    
    PWM0 POLARITY:  normal

streamnsight, 2016-09-25:
Compile new kernel, etc...

now I'm getting

```
>>> import CHIP_IO.OverlayManager as OM
>>> import CHIP_IO.PWM as PWM
>>> OM.load("PWM0")
>>> PWM.start("PWM0", 50)
ASSRT failed at source/c_pwm.c:92 (s == len)
```

Is this the code that's failing?

xtacocorex, 2016-09-25:
Yeah, that's the code that's telling us there is an issue.  Essentially, the check for writing to the sysfs file is failing.  The check looks to see if what was written matches the length of data supposed to be written.  The investigation into this showed me that the sysfs files for settings are open, just can only be written in a certain order that changes occasionally. 

I've pushed up a branch that has the asserts removed and has a ton of debug enabled.  https://github.com/xtacocorex/CHIP_IO/tree/hotfix/hwpwmfix

It still doesn't really work though as the issue is getting the data into sysfs.

I tried looking at this issue a little bit today as I got a question on SoftPWM and servo motors on the irc chat.  I did learn that I have no idea how the PWM stuff really works for controlling stuff (I mainly stick to i2c).

closed 2016-12-31 by xtacocorex