xtacocorex/CHIP_IO #18: Implement the all powerful CleanupAll()

xtacocorex, 2016-07-28:
Based on a suggestion by @aninternetof in Issue #16, we should probably do this as it seems super helpful.

aninternetof, 2016-07-28:
We might have to get creative when finding a way to tie this into [the Adafruit module](https://github.com/xtacocorex/Adafruit_Python_GPIO), since I don't think the RaspPi or BBB gpio modules have any feature like this.

xtacocorex, 2016-07-29:
So there is @fordsfords script already in the file structure.  It's currently not setup to install anywhere.  I can do that, but I have questions:

1. The script seems super generic, does it only do the XIO?
On a second glance, I think it'll do everything.

I already answered my other question of if it'll unexport the gpiochip0 and gpiochip1016, but I don't think it will.

This becomes tricky if you loaded a DTB overlay for PWM.  Dumping the overlay will kill the pwm export, but sure of the side effects of forced DTB overlay removal if the pwm0 is exported.

fordsfords, 2016-07-29:
I wasn't thinking that the script would get "installed" on a system using CHIP_IO, but I guess it wouldn't hurt.  No objection here.  I just had it as a ease for *developers* of the CHIP_IO package.

Yes, it knows to skip the "gpiochip*" entries.  And yes, it will unexport EVERYTHING, including CSI, LCD, etc, etc.

To kill only the XIOs would require a bit more sophistication, and would use some of the code from https://github.com/fordsfords/gpio_sh

xtacocorex, 2016-12-31:
I went ahead and added this as a Python function to the Utilities.py file.  Going to go ahead and close this.

closed 2016-12-31 by xtacocorex