xtacocorex/CHIP_IO #19: Overlay Manager fails

halherta, 2016-08-17:
I'm unable to get the overlay manager to work.

When i type the following in the REPL:
```
>>> import CHIP_IO.OverlayManager as OM
>>> OM.enable_debug()
>>> OM.load("PWM0")
```

I get the following error:
```
LOAD OVERLAY: PWM0 @ 
VALID OVERLAY
CONFIG PATH:  /sys/kernel/config/device-tree/overlays/chip-pwm
OVERLAY PATH: /lib/firmware/chip_io/chip-pwm0.dtbo
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.4/dist-packages/CHIP_IO-0.2.2-py3.4-linux-armv7l.egg/CHIP_IO/OverlayManager.py", line 180, in load
    errc = _set_overlay_verify(overlay.upper(), opath, cpath)
  File "/usr/local/lib/python3.4/dist-packages/CHIP_IO-0.2.2-py3.4-linux-armv7l.egg/CHIP_IO/OverlayManager.py", line 84, in _set_overlay_verify
    os.makedirs(config_path)
  File "/usr/lib/python3.4/os.py", line 227, in makedirs
    makedirs(head, mode, exist_ok)
  File "/usr/lib/python3.4/os.py", line 227, in makedirs
    makedirs(head, mode, exist_ok)
  File "/usr/lib/python3.4/os.py", line 237, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/sys/kernel/config/device-tree'
```

Any feedback would be appreciated. Thanks.

xtacocorex, 2016-08-17:
Questions:
1. Did you re-compile your kernel to enable CONFIG_OF_CONFIGFS?
2. Did you run the python interpreter with sudo?

If not, I'll have to look at it when I get back to my CHIP. I'm currently out of town.

halherta, 2016-08-17:
I did not recompile the Kernel to enable CONFIG_OF_CONFIGS. It seems like a logical thing to have enabled in the default kernel. 

Any tutorials on procedure to compile the Kernel for the CHIP? 
Thanks

xtacocorex, 2016-08-26:
This guide is good for cross compiling the kernel: http://www.chip-community.org/index.php/Compile_the_Linux_kernel_for_CHIP

Just have to change the git checkout commands to point to the 4.4.11 branch on github instead of 4.3.

When you get to the make menuconfig step, you'll need to enable the configfs stuff.

xtacocorex, 2016-09-22:
I put up my 4.4.13 kernel for people to use if they want to, it has the ability to load DTB Overlays: http://www.chip-community.org/index.php/4.4.13_LRADC_Test.

Going to close this issue.

closed 2016-09-22 by xtacocorex