xtacocorex/CHIP_IO #21: internal pull up/down resistor

davidbrock, 2016-08-27:
Does chip have internal pull up or down resistors that I can enable?  I would prefer this over using a bread board.  I have tried using
`GPIO.setup("CSID3", GPIO.IN, pull_up_down=GPIO.PUD_UP)`
I have also tried "XIO-P0".
I don't get any errors but the pin appears to be floating.

fordsfords, 2016-08-27:
Hi David.  I am not very familiar with SOCs that have software-controlled pullups, nor am I familiar with the details of the GPIO pins on the A8 CPU.  But the XIO does not have software controllable pullups.

That said, the XIO lines are not technically floating either.  There is a current source of about 50 uA.  So the output voltage depends on the load resistance.  Loads above 80K will have 3.3V across them.  Thus, an unconnected XIO pin will read as "1".  As the load resistance decreases, the voltage decreases to maintain a 50 uA current flow through the load.

davidbrock, 2016-09-01:
Thanks, that solved it. I now have buttons connected between ground and the XIO pins.

closed 2016-09-01 by davidbrock