xtacocorex/CHIP_IO #22: tests failing

andreibosco, 2016-09-10:
Running `sudo py.test` on a fresh install results in 9 failures:
```
============================= test session starts ==============================
platform linux2 -- Python 2.7.9 -- py-1.4.25 -- pytest-2.6.3
collected 63 items

test/test_gpio_input.py ..
test/test_gpio_output.py ......
test/test_gpio_setup.py ..........
test/test_pwm_setup.py FFFF...........F.........
test/test_softpwm_setup.py ...............softpwm_start failedFsoftpwm_start failedFsoftpwm_start failedFsoftpwm_start failedF.

=================================== FAILURES ===================================
_________________________ TestPwmSetup.test_start_pwm __________________________

self = <test_pwm_setup.TestPwmSetup instance at 0xb668fb70>

    def test_start_pwm(self):
        PWM.start("PWM0", 0)
    
        pwm_test = '/sys/class/pwm/pwmchip0/pwm0'
    
>       assert os.path.exists(pwm_test)
E       assert <function exists at 0xb6d80d30>('/sys/class/pwm/pwmchip0/pwm0')
E        +  where <function exists at 0xb6d80d30> = <module 'posixpath' from '/usr/lib/python2.7/posixpath.pyc'>.exists
E        +    where <module 'posixpath' from '/usr/lib/python2.7/posixpath.pyc'> = os.path

test/test_pwm_setup.py:15: AssertionError
________________ TestPwmSetup.test_start_pwm_with_polarity_one _________________

self = <test_pwm_setup.TestPwmSetup instance at 0xb67804b8>

    def test_start_pwm_with_polarity_one(self):
        PWM.start("PWM0", 0, 2000, 1)
    
        pwm_test = '/sys/class/pwm/pwmchip0/pwm0'
    
>       assert os.path.exists(pwm_test)
E       assert <function exists at 0xb6d80d30>('/sys/class/pwm/pwmchip0/pwm0')
E        +  where <function exists at 0xb6d80d30> = <module 'posixpath' from '/usr/lib/python2.7/posixpath.pyc'>.exists
E        +    where <module 'posixpath' from '/usr/lib/python2.7/posixpath.pyc'> = os.path

test/test_pwm_setup.py:27: AssertionError
______________ TestPwmSetup.test_start_pwm_with_polarity_default _______________

self = <test_pwm_setup.TestPwmSetup instance at 0xb6780be8>

    def test_start_pwm_with_polarity_default(self):
        PWM.start("PWM0", 0, 2000, 0)
    
        pwm_test = '/sys/class/pwm/pwmchip0/pwm0'
    
>       assert os.path.exists(pwm_test)
E       assert <function exists at 0xb6d80d30>('/sys/class/pwm/pwmchip0/pwm0')
E        +  where <function exists at 0xb6d80d30> = <module 'posixpath' from '/usr/lib/python2.7/posixpath.pyc'>.exists
E        +    where <module 'posixpath' from '/usr/lib/python2.7/posixpath.pyc'> = os.path

test/test_pwm_setup.py:41: AssertionError
________________ TestPwmSetup.test_start_pwm_with_polarity_zero ________________

self = <test_pwm_setup.TestPwmSetup instance at 0xb66bd698>

    def test_start_pwm_with_polarity_zero(self):
        PWM.start("PWM0", 0, 2000, 0)
    
        pwm_test = '/sys/class/pwm/pwmchip0/pwm0'
    
>       assert os.path.exists(pwm_test)
E       assert <function exists at 0xb6d80d30>('/sys/class/pwm/pwmchip0/pwm0')
E        +  where <function exists at 0xb6d80d30> = <module 'posixpath' from '/usr/lib/python2.7/posixpath.pyc'>.exists
E        +    where <module 'posixpath' from '/usr/lib/python2.7/posixpath.pyc'> = os.path

test/test_pwm_setup.py:55: AssertionError
_____________________ TestPwmSetup.test_pwm_duty_modified ______________________

self = <test_pwm_setup.TestPwmSetup instance at 0xb66bdaf8>

    def test_pwm_duty_modified(self):
        PWM.start("PWM0", 0)
    
        pwm_test = '/sys/class/pwm/pwmchip0/pwm0'
    
>       assert os.path.exists(pwm_test)
E       assert <function exists at 0xb6d80d30>('/sys/class/pwm/pwmchip0/pwm0')
E        +  where <function exists at 0xb6d80d30> = <module 'posixpath' from '/usr/lib/python2.7/posixpath.pyc'>.exists
E        +    where <module 'posixpath' from '/usr/lib/python2.7/posixpath.pyc'> = os.path

test/test_pwm_setup.py:115: AssertionError
_________ TestSoftpwmSetup.test_pwm_duty_cycle_invalid_value_negative __________

self = <test_softpwm_setup.TestSoftpwmSetup instance at 0xb66a4238>

    def test_pwm_duty_cycle_invalid_value_negative(self):
>       PWM.start("XIO-P7", 0)
E       RuntimeError: Error starting softpwm on pin U14_20 (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)))

test/test_softpwm_setup.py:85: RuntimeError
__________ TestSoftpwmSetup.test_pwm_duty_cycle_invalid_value_string ___________

self = <test_softpwm_setup.TestSoftpwmSetup instance at 0xb66a48a0>

    def test_pwm_duty_cycle_invalid_value_string(self):
>       PWM.start("XIO-P7", 0)
E       RuntimeError: Error starting softpwm on pin U14_20 (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy))
E       gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy))
E       gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)))

test/test_softpwm_setup.py:91: RuntimeError
__________ TestSoftpwmSetup.test_pwm_frequency_invalid_value_negative __________

self = <test_softpwm_setup.TestSoftpwmSetup instance at 0xb66b6080>

    def test_pwm_frequency_invalid_value_negative(self):
>       PWM.start("XIO-P7", 0)
E       RuntimeError: Error starting softpwm on pin U14_20 (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy))
E       gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy))
E       gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy))
E       gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (g)

test/test_softpwm_setup.py:97: RuntimeError
___________ TestSoftpwmSetup.test_pwm_frequency_invalid_value_string ___________

self = <test_softpwm_setup.TestSoftpwmSetup instance at 0xb66551e8>

    def test_pwm_frequency_invalid_value_string(self):
>       PWM.start("XIO-P7", 0)
E       RuntimeError: Error starting softpwm on pin U14_20 (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy))
E       gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy))
E       gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy))
E       gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (gpio_export: could not write '1023' to /sys/class/gpio/export (Device or resource busy)
E       Error setting up softpwm on pin 1023, maybe already exported? (g)

test/test_softpwm_setup.py:103: RuntimeError
===================== 9 failed, 54 passed in 2.12 seconds ======================
```

xtacocorex, 2016-09-10:
Looks like you're running a 4.4 CHIP.

For HW PWM, you have to have either PWM0 setup in the stock DTB, or a custom kernel that enables CONFIG_OF_CONFIGFS to allow for DTB Overlays.  PWM0 isn't set up by default on 4.4 yet.

As for Soft PWM, it looks like the issue is related to a GPIO already exported.  I'll look into it.

xtacocorex, 2016-12-31:
Made the PWM test "skip" over 4 of the failing tests.

xtacocorex, 2016-12-31:
Fixed the SoftPWM pytest code.

closed 2016-12-31 by xtacocorex