xtacocorex/CHIP_IO #24: no luck with HW PWM0

streamnsight, 2016-09-23:
I got SOFTPWM working fine, except it doesn't seem to be very precise if I can trust my oscilloscope: asking for 50z I get something more like 45Hz, and can never set duty cycle down lower than ~5%

So I wanted to try HW PWM, and hooked things up on PWM0, but no response.
Using only PWM.start("PWM0", 5, 50), no error but no response.

So I thought maybe I need to activate PWM0 overlay.

I ran 

```
>>> import CHIP_IO.OverlayManager as OM
>>> OM.load("PWM0")
```
and I am getting:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "build/bdist.linux-armv7l/egg/CHIP_IO/OverlayManager.py", line 180, in load
  File "build/bdist.linux-armv7l/egg/CHIP_IO/OverlayManager.py", line 84, in _set_overlay_verify
  File "/usr/lib/python2.7/os.py", line 150, in makedirs
    makedirs(head, mode)
  File "/usr/lib/python2.7/os.py", line 150, in makedirs
    makedirs(head, mode)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 1] Operation not permitted: '/sys/kernel/config/device-tree'
```

I'm running python with `sudo` so not sure what else I need to do.
I installed the dtb as per the instructions.

Any ideas?
Thanks for help.

xtacocorex, 2016-09-23:
Are you running a custom 4.4 kernel with CONFIG_OF_CONFIGFS enabled in the kernel build?  The stock CHIP 4.4 kernel doesn't have this enabled, so overlays don't work out of the box.

I uploaded my 4.4.13 test kernel and the instructions for installation are here: http://www.chip-community.org/index.php/4.4.13_LRADC_Test

Not sure if you saw Issue #17, but HW PWM has issues at the moment that I haven't yet resolved.

SoftPWM has it's quirks and isn't meant for high speed work as it just converts a standard GPIO pin and toggles it on/off on set times.

streamnsight, 2016-09-23:
@xtacocorex 

Thanks for the quick reply.

I'm running a custom 4.4.11 kernel I built to support Docker that requires Overlays, so I think it's in there.

The config is here: 
https://gist.github.com/streamnsight/017d9bc708a1b34d9034c483454bca45

I don't see a `CONFIG_OF_CONFIGFS` there but there is 
```
CONFIG_PWM=y
CONFIG_PWM_SYSFS=y
CONFIG_PWM_SUN4I=y
CONFIG_OVERLAY_FS=y
```

Is that what you mean?

With regards to https://github.com/xtacocorex/CHIP_IO/issues/17 , I'm not sure to understand what's actually broken, so I guess I would not be of much help to fix it, but if there is a way I can help, let me know.

Thanks

streamnsight, 2016-09-23:
ahhh.. sorry for my ignorance. I grep'ed for commented out flags and `CONFIG_OF_CONFIGFS` is not enabled.

On to recompiling a kernel I guess.

streamnsight, 2016-09-23:
compiled a new kernel, and took the opportunity to add the UART2 as per 
https://bbs.nextthing.co/t/second-serial-port/4163/16

Will test tonight.

xtacocorex, 2016-09-23:
From your adding of UART2, do you think that's something easily added via an overlay?  I just bookmarked that thread in the hopes of diving into that sometime here.

streamnsight, 2016-09-23:
well, the dts and dtsi files need to be edited, but also a driver file `drivers/pinctrl/sunxi/pinctrl-sun5i-a13.c` to add uart function: 
So that means a patch to the code, but as long as it's implemented in future kernels, i think it's totally doable.

BTW, I very much appreciate your work on this; makes life a lot easier.

I'm surprised that for a platform like CHIP, which is very much suited for IoT applications, there is so little support for I/Os out of the box. It feels like it was built as a mini computer with a nice UI for educational purposes, but it has so much more potential, if only it was made easier to get working with.

xtacocorex, 2016-09-23:
Just have to remember it's in its infancy and we're on the ground floor of getting things setup.  :)
I think NTC has done the process right though, get OS drops out for us to hack up.  Does it suck a little bit, yeah, but it's not terrible to deal with.  I had to do a lot of work when 4.4 came out update this and it required the help of @fordsfords and @aninternetof to get it hashed out (and HW PWM broke in the process).  In the end, we have a fairly complete library for everyone to use.

The only reason CHIP_IO exists is because I needed it for my projects and at the time, the only Python IO control was in the Chippy Ruxpin repo.

Thanks for your feedback and comments on this, I appreciate it and it makes me happy that something that I've needed can be useful to everyone else.

xtacocorex, 2016-12-31:
In reading through this again, it appears that this is actually resolved and was due to the kernel not having CONFIG_OF_CONFIGFS enabled.  Going to close this.

closed 2016-12-31 by xtacocorex