xtacocorex/CHIP_IO #25: PWM lines

streamnsight, 2016-09-23:
just a question here:

I just realize there only seem to be one PWM pin, (PWM0)
Can other pins be used for HW PWM, or is that it?

'coz I was looking to control a few motors, but if there is only one pin that's not going to work! :-/

aninternetof, 2016-09-23:
Hello @streamnsight! You are correct -- only one hardware PWM pin.
But, you can use lots of soft PWMs! See the SOFTPWM section of the readme. They are a little less precise than than the hardware PWM, but I've been using them to control some motors and haven't had any problems.

streamnsight, 2016-09-23:
@aninternetof 

Thanks for the quick note.
I'm trying to use PWM as PPM, so precision of at least 100us is key.

I tried SOFTPWM on `XIO-P7` and `XIO-P5` and when requesting 50Hz I get ~45Hz.

I couldn't go any lower than 6-7% duty cycle at the min duty cycle setting (~1.4ms @ 50Hz)
I'm not sure why, but if I can tune the frequency, duty cycle limitations like this are a no-go:
For PPM, I need to tune between 5% and 10% duty cycle (1ms to 2ms @ 50Hz), so 1.4ms is the middle of the range. :-/

What's your experience with motors and PWM? what kind of frequency / duty cycle are you running at?

I think I might just go with a i2c PWM controller, and I can get down to 2us precision at 60Hz.
Might be a better option than fiddle with PWM

aninternetof, 2016-09-23:
@streamnsight Thanks for the details. With your precision needs, an i2c PWM sounds like it could be your best option.

My experience with using this SOFTPWM module with motors required much less precision. I run at about 100Hz, but I'm only looking for approximate duty cycles like 10%, 20%, 30%, etc.

When I was experimenting with the SOFTPWM during development, I saw the limitations at the very low duty cycles just like you are seeing. Unfortunately, it's very hard to get much less than 20ms precision out of a software-based PWM running on top of a big OS like Linux.

Let us know if you have any trouble setting up an i2c PWM controller! I know @xtacocorex has been working very hard on device-tree overlays to make sure that i2c is easy to use on every CHIP.

aninternetof, 2016-09-23:
Although, at only 50Hz, seeing 45% when you requested 50% seems worse than I expected. I'll try to test this out.

xtacocorex, 2016-11-05:
@streamnsight sorry for not working on this for a while, I got super busy and have had my CHIP powered off for quite some time.  I noticed you forked the code and have done some mods to the PWM stuff; have you gotten anything working?

streamnsight, 2016-11-05:
actually no, also got side tracked trying to get the cedrus VPU driver going on a 4.4 kernel and got stuck there too...

getting a little frustrated with the DIY part of CHiP. :-/

xtacocorex, 2016-11-05:
Understandable.  I need to see what the unified kernel does with the CHIP when NTC drops that image for us.

streamnsight, 2016-11-05:
do you happen to know if they keep a repo of patches i could apply to the mainline kernel? especially the NAND patch to boot from flash. 

the cedrus driver is in mainline, but when i tried to backport i ran into so many dependency issues i gave up. i tried porting the NAND stuff to mainline but i am not sure what's involved so got stuck. 

would be nice if thete was a set of patches to apply to a core kernel to make it run on CHIP.

xtacocorex, 2016-11-05:
So there isn't an easy way, but you can compare branches on github.  The link below compares debian/4.4.11-ntc-1 with debian/4.4.13-ntc-mlc

https://github.com/NextThingCo/CHIP-linux/compare/debian/4.4.11-ntc-1...debian/4.4.13-ntc-mlc

xtacocorex, 2016-12-31:
@aninternetof, I'm looking into any possible issues with SoftPWM at the moment (in an effort to close this out) and it doesn't seem to work on the CSI pins. (I'm guessing the LCD pins won't work either since their direct to the R8 also) 

Is this the case for you?  

I'm driving a 180deg JR Model Airplane Servo, powered from the 3V3 pin on U13.  It definitely works on XIO-P7, and therefore will work on the other XIO pins.

aninternetof, 2016-12-31:
@xtacocorex -- I don't think I tested with anything but the XIO lines, so I'll try out a CSI pin when I'm back at my desk in Monday!

xtacocorex, 2016-12-31:
No worries on time frame, was more-so curious.  I did confirm the LCD pins also don't work for me.  I started some code (that has been checked in) that adds an identifier to the pins if it can be used for SoftPWM.  I haven't added any checking code to use that information when trying to setup SoftPWM as I don't want to start adding code that prevents it from working for other people if my servo is the issue.

xtacocorex, 2017-01-29:
I'm going to go ahead and close this issue.  If the R8 GPIO cannot be used for SoftPWM, we can create another issue to dis-allow that ability.

closed 2017-01-29 by xtacocorex