xtacocorex/CHIP_IO #26: Touchscreen ADC-s

laurivosandi, 2016-10-10:
Hello, as far as I know there are 4 ADC-s (X1, X2, Y1, Y2) for the touchscreen. Would it be possible for you to expose these via the CHIP-IO library?

xtacocorex, 2016-11-05:
The current plan is to get LRADC working with CHIP_IO first.  Once I get that done, I can attempt to start on the touchscreen ADC's.  I'm not sure they're broken out to sysfs, so it may be super tricky.

xtacocorex, 2016-12-31:
@laurivosandi, is this the only thing that needs to happen to get the GPADC working on the CHIP? https://github.com/plaes/linux/commit/3c59759887468e3c5795cee8c2e32885ae516bf7

I have a feeling that having both CONFIG_TOUCHSCREEN_SUN4I set (for the combined OS that works with the PocketCHIP) along with CONFIG_MFD_SUN4I_GPADC and CONFIG_SUN4I_GPADC might cause issues.

I'll go throw an issue up on the CHIP Kernel repo to see what those guys say.

xtacocorex, 2016-12-31:
Referencing: https://github.com/NextThingCo/CHIP-linux/issues/72

xtacocorex, 2018-12-29:
Due to this requiring a custom kernel and the CHIP being a dead end product, this issue will be closed.

closed 2018-12-29 by xtacocorex