xtacocorex/CHIP_IO #27: aREST support

dtraub, 2016-11-05:
More of a feature request than an issue.

Any way you can add aREST support for this?  I am not the greatest at programming, but I feel it should be possible as its very close to a RPi.

xtacocorex, 2016-11-05:
In my quick search of aREST, it's written in node.js.  I don't have much experience with javascript.  I also can't find any docs on what the API looks like in order to mimic it in Python.  It seems that everything is behind a paywall. :(

dtraub, 2016-11-05:
Bummer, thanks for looking into it!

xtacocorex, 2016-11-05:
I did dig around some more (the benefits of being sick and just sitting around doing nothing) and found this: https://github.com/marcoschwartz/pi-aREST

That looks to be the server that runs and accepts the commands.

I'll see what I can do, can't guarantee it'll be quickly done though.

dtraub, 2016-11-05:
thanks, is this basically it?

https://github.com/sandeepmistry/node-chip-io

xtacocorex, 2016-12-31:
The node-chip-io code is Node.js support for controlling the GPIO, I don't believe there is anything in there for REST like APIs.

xtacocorex, 2017-01-01:
So I've started working on this @dtraub. I'm debating on whether or not it should go into its own repo or keep it in here.  Thoughts?

dtraub, 2017-01-01:
Personally, it would probably be better to have in a separate repo due to the specific use cases.  Thanks again!

xtacocorex, 2017-01-02:
@dtraub I'll start a new repo for it.  Going to close this issue as the repo is live, but no actual code has been uploaded yet.  https://github.com/xtacocorex/chip-python-aREST

closed 2017-01-02 by xtacocorex