xtacocorex/CHIP_IO #28: Install failed

benmarten, 2016-11-24:
I'm relatively new to python, and I got chip working, but after reboot its not working anymore:

```
chip@chip ~/dev> sudo python3 test.py 
[sudo] password for chip: 
Error processing line 1 of /usr/local/lib/python3.4/dist-packages/easy-install.pth:

  Traceback (most recent call last):
    File "/usr/lib/python3.4/site.py", line 177, in addpackage
      if not dircase in known_paths and os.path.exists(dir):
    File "/usr/lib/python3.4/genericpath.py", line 19, in exists
      os.stat(path)
  TypeError: embedded NUL character

Remainder of file ignored
Traceback (most recent call last):
  File "test.py", line 1, in <module>
    import CHIP_IO.GPIO as GPIO
ImportError: No module named 'CHIP_IO'
```

I'm wondering why chip_io isn't available by a package manager?

benmarten, 2016-11-24:
Rerunning setup fails as well:

```
chip@chip ~/d/CHIP_IO> sudo python3 setup.py install
Error processing line 1 of /usr/local/lib/python3.4/dist-packages/easy-install.pth:

  Traceback (most recent call last):
    File "/usr/lib/python3.4/site.py", line 177, in addpackage
      if not dircase in known_paths and os.path.exists(dir):
    File "/usr/lib/python3.4/genericpath.py", line 19, in exists
      os.stat(path)
  TypeError: embedded NUL character

Remainder of file ignored
Compiling DTS Files
running install
Checking .pth file support in /usr/local/lib/python3.4/dist-packages/
/usr/bin/python3 -E -c pass
Error processing line 1 of /usr/local/lib/python3.4/dist-packages/easy-install.pth:

  Traceback (most recent call last):
    File "/usr/lib/python3.4/site.py", line 177, in addpackage
      if not dircase in known_paths and os.path.exists(dir):
    File "/usr/lib/python3.4/genericpath.py", line 19, in exists
      os.stat(path)
  TypeError: embedded NUL character

Remainder of file ignored
TEST PASSED: /usr/local/lib/python3.4/dist-packages/ appears to support .pth files
Traceback (most recent call last):
  File "setup.py", line 35, in <module>
    Extension('CHIP_IO.SOFTPWM', ['source/py_softpwm.c', 'source/c_softpwm.c', 'source/constants.c', 'source/common.c', 'source/event_gpio.c'], extra_compile_args=['-Wno-format-security'])]) #,
  File "/usr/lib/python3.4/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.4/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.4/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 103, in do_egg_install
    cmd.ensure_finalized()  # finalize before bdist_egg munges install cmd
  File "/usr/lib/python3.4/distutils/cmd.py", line 107, in ensure_finalized
    self.finalize_options()
  File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 323, in finalize_options
    self.check_site_dir()
  File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 455, in check_site_dir
    self.pth_file = PthDistributions(pth_file, self.all_site_dirs)
  File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 1531, in __init__
    self._load()
  File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 1553, in _load
    os.path.join(self.basedir, path)
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2052, in normalize_path
    return os.path.normcase(os.path.realpath(filename))
  File "/usr/lib/python3.4/posixpath.py", line 372, in realpath
    path, ok = _joinrealpath(filename[:0], filename, {})
  File "/usr/lib/python3.4/posixpath.py", line 406, in _joinrealpath
    if not islink(newpath):
  File "/usr/lib/python3.4/posixpath.py", line 161, in islink
    st = os.lstat(path)
TypeError: embedded NUL character
```

Any ideas?

xtacocorex, 2016-11-24:
Gotta standby a bit, so much food in my belly from Thanksgiving and the stout I've had to drink isn't helping anything.

The test.py script has to be run from within the CHIP_IO directory and there will most likely be issues with some of the tests.

You do need the DTC compiler, that NTC provides, installed before installing CHIP_IO.

I don't personally use Python3, so there most definitely could be an install issue.  I'll have to get my dev CHIP up and running when I get home to test it out.

As for the the package manager, it can be installed via pip since I uploaded a version there (can't remember which one though). 
It was mentioned by @computermouth that it could possibly be added to the CHIP repo, so that could happen in the future.

xtacocorex, 2016-11-25:
Ok, back home and looked over stuff.  I just installed CHIP_IO under Python 2.7.9 and Python 3.4 on a freshly flashed headless CHIP.  No errors in either install.

I rebooted the CHIP and tried the Python3 install again.  No issues.

What does your test.py script look like?

benmarten, 2016-11-25:
So i got it working with python2.7. I reran the whole install instructions again, and it installed the dtc dependencies again... prob. that might have been missing. Anyways its working now. Thanks for checking!

closed 2016-11-25 by benmarten

xtacocorex, 2016-11-29:
Glad it worked out!