xtacocorex/CHIP_IO #29: Minor issue building dtc

gpsbabel, 2016-11-28:
Perhaps it's a missing dependency upstream, but the make of dtc fails as version_gen.h is not provided in the dtc repo.  A 'make version_gen.h' is required before the make of dtc itself.  Thanx!

xtacocorex, 2016-12-01:
Is this using the dtc-compiler from NTC?  I haven't built DTC in a while, the ones with CHIP_IO also have the binary files that are installed via setup.py.

gpsbabel, 2016-12-01:
It's using the dtc from "atenarc" as per
https://github.com/xtacocorex/CHIP_IO So whether the issues is in the
atenarc dithub Makefile or your instructions is your call. I could see it
going either way.  Thanx.

xtacocorex, 2016-12-30:
In setting up a CHIP specifically for CHIP_IO dev, I forgot to setup the DTC compiler.  I didn't have to run the make version_gen.h when I got it all setup.  Will investigate this more once I get the PWM stuff figured out.

xtacocorex, 2017-02-13:
I'm removing the DTB compilation requirement from CHIP_IO. OverlayManager will handle the PWM and SPI sample dtb from chip-dt-overlays.  Going to close this.

closed 2017-02-13 by xtacocorex