xtacocorex/CHIP_IO #32: Handle CHIP Pro

xtacocorex, 2016-12-01:
The NTC CHIP Pro is a smaller CHIP with the GR8 SOC that is designed for embedded applications.

The IO of the CHIP Pro is slightly different from the CHIP.

Need to come up with an elegant way to handle IO on the CHIP Pro while maintaining compatibility with the CHIP/PocketCHIP.

Flexshot, 2017-01-12:
Would love to discuss how this should be implemented. I have a method in mind and most likely so do you. Feel free to message me in the forums. I can implement this in a few days.

xtacocorex, 2017-01-13:
I really didn't have a method in mind just yet, was moreso going to try it on the CHIP Pro (something you're already doing) and see where things failed miserably.  

I was going to add another field to the pins_t struct that denotes CHIP/CHIP Pro support.  The XIOs and most of the LCD pins are definitely not on the CHIP Pro, but I haven't dove into and really researched everything yet.

xtacocorex, 2017-01-29:
Add per PWM channel cleanup since there are 2 PWM channels on the CHIP Pro.  This would help chip-python-aREST...

xtacocorex, 2017-02-15:
Forgot to add the issue tag to this commit: https://github.com/xtacocorex/CHIP_IO/commit/c5dbede3c4cd37552f37af7f5396fd1c5af72a87

commit bab94f76 (2017-02-26): forgot the actual version bump for #32. gonna close #32 since it's initially done
closed 2017-02-26 by xtacocorex