xtacocorex/CHIP_IO #34: Prevent file opening whenever GPIO value is set

laurivosandi, 2016-12-11:
Currently it seems file is opened whenever GPIO value is set (eg from soft PWM). I think it would make more sense to open the file handle when pin is exported and close the file handle when pin is unexported.

https://github.com/xtacocorex/CHIP_IO/blob/master/source/event_gpio.c#L385

xtacocorex, 2016-12-30:
So, the Hardware PWM code has this feature and I am beginning to wonder if it's the reason why the code isn't working properly.  I'm still investigating all the things at the moment, but I'll keep this under consideration.

laurivosandi, 2016-12-30:
With current sysfs approach it's not going to get much better but opening/closing file handle every PWM cycle seems rather silly :)

xtacocorex, 2016-12-31:
This was actually a whole lot easier than I was thinking.  My thought process was over complicating things.

closed 2016-12-31 by xtacocorex

laurivosandi, 2017-01-01:
Nice job, going to give a try tomorrow :)

commit 9a2e81f0 (2016-12-31): fix to close #34, gpio value file is now opened as read/write instead of read only