xtacocorex/CHIP_IO #36: Add way to easily and quickly toggle pin direction

xtacocorex, 2016-12-16:
Per discussion here: https://bbs.nextthing.co/t/change-gpio-direction/13162

blueSolder, 2016-12-24:
I have gone over the source code and think that direction changing can be implemented by copying  py_setup_channel in py_gpio.c rename copy, edit out not required lines and add a new entry to gpio_methods[].
Not sure what to do next though, am new to github.

xtacocorex, 2016-12-24:
Yeah, it shouldn't be too hard to implement this feature. Just need to get past the next 2 days and I'll have some free time to work on CHIP_IO.

blueSolder, 2016-12-25:
Have done above and got it working.

aninternetof, 2016-12-25:
That's great! If you'd like to make a pull request, that would be one way to let us try out and adopt your change.

xtacocorex, 2016-12-25:
I second what @aninternetof said @blueSolder.  If you forked the code, you can definitely contribute and submit a pull request. Otherwise, I plan on getting to that this week.

xtacocorex, 2016-12-26:
Closed per PR#39 being merged.

closed 2016-12-26 by xtacocorex