xtacocorex/CHIP_IO #38: integration with PureData?

batchku, 2016-12-22:
Thank you for this wonderful library; awesome contribution.

I'd like to control the GPIO pins of the CHIP from PureData (http://puredata.info/).
Any advice on this integration? #

xtacocorex, 2016-12-24:
So I did some brief research on PureData and, not going to lie, it seems super confusing to even figure out what it's supposed to do.

Is it a more advanced version of Scratch? Everything I found made it seem like it's for audio stuff.

jtrip, 2017-05-06:
PureData is like a _backend_ for music systhesis software, you can work with it directly or there are other popular tools. It's pretty cool, I don't know all the details but it is really popular with electronic musicians.