xtacocorex/CHIP_IO #40: SPI fails to load - sun4i-spi 1c17000.spi: prop pinctrl-0 index 0 invalid phandle

stokmanis, 2016-12-31:
I am using CHIP version 4.4.13-ntc-mlc #1 SMP Tue Dec 6 21:38:00 UTC 2016 armv7l GNU/Linux
when running
```
import CHIP_IO.OverlayManager as OM
OM.load("SPI2")
```

in the console I am getting
sun4i-spi 1c17000.spi: prop pinctrl-0 index 0 invalid phandle
sun4i-spi 1c17000.spi: chipselect 0 already in use
spi_master spi32766: spi_device register error /soc@01c00000/spi@01c17000/spi2@0
of_spi_notify: failed to create for '/soc@01c00000/spi@01c17000/spi2@0'
__of_changeset_entry_notify: notifier error @/soc@01c00000/spi@01c17000/spi2@0

Any ideas how to resolve it?

xtacocorex, 2016-12-31:
What happens if you do an:

    ls /dev/sp*

When I load the SPI2 overlay (manually in the Python prompt):

    [chip@chipdev][15:07][test]: sudo python
    Python 2.7.9 (default, Aug 13 2016, 17:56:53)
    [GCC 4.9.2] on linux2
    Type "help", "copyright", "credits" or "license" for more information.
    >>> import CHIP_IO.OverlayManager as om
    >>> om.load("SPI2")
    >>> om.load("SPI2")
    SPI2 Overlay already loaded
    2
    >>>

I get the following device in /dev:

    /dev/spidev32766.0

I am 98% sure that things will be ok if you get the actual SPI device in /dev/.

When I run dmesg, in another terminal instance, I do get the following (the same error as yourself):

    [81453.310000] sun4i-spi 1c17000.spi: prop pinctrl-0 index 0 invalid phandle
    [81453.345000] sun4i-spi 1c17000.spi: chipselect 0 already in use
    [81453.350000] spi_master spi32766: spi_device register error /soc@01c00000/spi@01c17000/spi2@0
    [81453.360000] of_spi_notify: failed to create for '/soc@01c00000/spi@01c17000/spi2@0'
    [81453.365000] __of_changeset_entry_notify: notifier error
@/soc@01c00000/spi@01c17000/spi2@0

Does the console output you're getting happen in the same instance as the Python script?  If so, that is odd and might be tricky to re-create on my end.

I believe the error is actually caused by the DTBO file, in how I created it.  @KoljaWindeler has reported the same issues in the CHIP BBS when he was working on creating overlays for his custom DIPs.  The DTB Overlay wants a specific device setup for SPI instead of just declaring spidev in it (which is what I have).  

I don't have an actual SPI device to test with, I should probably get one to validate one could actually communicate with it via python-spidev and my custom overlay loaded.

stokmanis, 2017-01-02:
Thanks for support. Yes, I am getting device file /dev/spidev32766.0 after om.load("SPI2").
Yes, you are correct - errors "sun4i-spi 1c17000.spi: prop pinctrl-0 index 0 invalid phandle" are in dmesg, not in the console where python script is run.

I have attached Logical Analyzer on pins CSICK (spi clk) and CSIPCK (spi cs), and run test script
```
wget http://www.raspibo.org/renzo/spidev_test.c 
gcc -o spidev_test spidev_test.c
sudo ./spidev_test -D /dev/spidev32766.0 
```
There are no signals

One idea is that the error "chipselect 0 already in use" somehow suggests that some other overlay has already reserved pin chipselect pin CSIPCK.

jpiat, 2017-01-18:
Hi,

i have the same issue here. Any idea on how to fix ? I have the exact same problem when i try to load the spi overlay provided with the base debian image.

stokmanis, 2017-01-27:
Resolved issue by forcing CHIP_IO OverlayManager to use NextThingCo sample-spi.dtbo file.
`sudo cp /lib/firmware/nextthingco/chip/sample-spi.dtbo /lib/firmware/chip_io/chip-spi2.dtbo`
Still when running OM.load("SPI2") there is even more garbage in dmesg - the same as at the end of this post https://isojed.nl/blog/2016/12/13/c-h-i-p-part-2/, but spi actually works.

Very good tip from https://bbs.nextthing.co/t/spi-serial-communication-on-chip/11937/11 and https://isojed.nl/blog/2016/12/13/c-h-i-p-part-2/ is to enable SPI by default by adding two lines in /etc/rc.local
```
mkdir -p /sys/kernel/config/device-tree/overlays/spi
cat /lib/firmware/nextthingco/chip/sample-spi.dtbo > /sys/kernel/config/device-tree/overlays/spi/dtbo
```

xtacocorex, 2017-01-28:
Thanks for the info and troubleshooting @stokmanis, I appreciate it.  I see what the difference is between mine and NTC's SPI overlay.  In an effort to not blatantly copy @mripards work, I decided to re-create the SPI overlay using his as a basis.  

My current plan to fix this is to remove my SPI overlay, change the code to point to the NTC created one, and update the README to include the chip-dt-overlays package in the install.

xtacocorex, 2017-01-28:
Should be fixed in this commit: https://github.com/xtacocorex/CHIP_IO/commit/eafcf0bf69e7b044f017eebcf81a879e5a2aef88
Forgot to properly tag the issue in the commit message. :(

closed 2017-01-28 by xtacocorex