xtacocorex/CHIP_IO #42: running tests fails with python2.7 and python3

nio101, 2017-01-02:
1) The instructions given for running tests are only relevant for python2.7.
2) `sudo py.test` fails, and `sudo python2 -m pytest` fails too:
```
========================================================================== test session starts ==========================================================================
platform linux2 -- Python 2.7.9 -- py-1.4.25 -- pytest-2.6.3
collected 0 items / 5 errors 

================================================================================ ERRORS =================================================================================
_______________________________________________________________ ERROR collecting test/test_gpio_input.py ________________________________________________________________
test/test_gpio_input.py:5: in <module>
    import CHIP_IO.GPIO as GPIO
E   ImportError: No module named GPIO
_______________________________________________________________ ERROR collecting test/test_gpio_output.py _______________________________________________________________
test/test_gpio_output.py:3: in <module>
    import CHIP_IO.GPIO as GPIO
E   ImportError: No module named GPIO
_______________________________________________________________ ERROR collecting test/test_gpio_setup.py ________________________________________________________________
test/test_gpio_setup.py:4: in <module>
    import CHIP_IO.GPIO as GPIO
E   ImportError: No module named GPIO
________________________________________________________________ ERROR collecting test/test_pwm_setup.py ________________________________________________________________
test/test_pwm_setup.py:5: in <module>
    import CHIP_IO.PWM as PWM
E   ImportError: No module named PWM
______________________________________________________________ ERROR collecting test/test_softpwm_setup.py ______________________________________________________________
test/test_softpwm_setup.py:4: in <module>
    import CHIP_IO.SOFTPWM as PWM
E   ImportError: No module named SOFTPWM
======================================================================== 5 error in 1.23 seconds ========================================================================
```
3) installing pytest with pip3/python3 and running `sudo python3 -m pytest` fails too... :(
```
========================================================================== test session starts ==========================================================================
platform linux -- Python 3.4.2, pytest-3.0.5, py-1.4.32, pluggy-0.4.0
rootdir: /home/chip/CHIP_IO, inifile: 
collected 0 items / 6 errors 

================================================================================ ERRORS =================================================================================
__________________________________________________________________ ERROR collecting test/lradc_test.py __________________________________________________________________
test/lradc_test.py:11: in <module>
    ADC.setup(125)
CHIP_IO/LRADC.py:57: in setup
    set_sample_rate(rate)
CHIP_IO/LRADC.py:116: in set_sample_rate
    print("Setting Sample Rate to: {0}").format(rate)
E   AttributeError: 'NoneType' object has no attribute 'format'
---------------------------------------------------------------------------- Captured stdout ----------------------------------------------------------------------------
ENABLING LRADC DEBUG OUTPUT
LRADC SETUP WITH SAMPLE RATE OF 125
Setting Sample Rate to: {0}
_______________________________________________________________ ERROR collecting test/test_gpio_input.py ________________________________________________________________
ImportError while importing test module '/home/chip/CHIP_IO/test/test_gpio_input.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
test/test_gpio_input.py:5: in <module>
    import CHIP_IO.GPIO as GPIO
E   ImportError: No module named 'CHIP_IO.GPIO'
_______________________________________________________________ ERROR collecting test/test_gpio_output.py _______________________________________________________________
ImportError while importing test module '/home/chip/CHIP_IO/test/test_gpio_output.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
test/test_gpio_output.py:3: in <module>
    import CHIP_IO.GPIO as GPIO
E   ImportError: No module named 'CHIP_IO.GPIO'
_______________________________________________________________ ERROR collecting test/test_gpio_setup.py ________________________________________________________________
ImportError while importing test module '/home/chip/CHIP_IO/test/test_gpio_setup.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
test/test_gpio_setup.py:4: in <module>
    import CHIP_IO.GPIO as GPIO
E   ImportError: No module named 'CHIP_IO.GPIO'
________________________________________________________________ ERROR collecting test/test_pwm_setup.py ________________________________________________________________
ImportError while importing test module '/home/chip/CHIP_IO/test/test_pwm_setup.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
test/test_pwm_setup.py:5: in <module>
    import CHIP_IO.PWM as PWM
E   ImportError: No module named 'CHIP_IO.PWM'
______________________________________________________________ ERROR collecting test/test_softpwm_setup.py ______________________________________________________________
ImportError while importing test module '/home/chip/CHIP_IO/test/test_softpwm_setup.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
test/test_softpwm_setup.py:4: in <module>
    import CHIP_IO.SOFTPWM as PWM
E   ImportError: No module named 'CHIP_IO.SOFTPWM'
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 6 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
======================================================================== 6 error in 1.89 seconds ========================================================================
```

xtacocorex, 2017-01-03:
So I just fixed some issues regarding the pytest code the other day, so I know that they work for Python2.7.  Well they used to, now that I did everything you've done, things aren't working right.  Blargh!  I'm not a Python3 guy and in my quick check I do get errors when trying:

    sudo python3 -m pytest

and they relate to string formatting (which is one of the reasons I haven't fully moved to Python3).

You are correct that I don't have instructions for running with Python3.  I can update those accordingly.  

I did notice that the LRADC test is running when using python3, it shouldn't be as it's not pytest formatted, so I'll fix that one.  Will have to figure this out now.

nio101, 2017-01-03:
There's no hurry for the tests. As to me, I'll stick to python 2.7 for now to use the module.
Thanks for you work! :)

commit 29377cf2 (2017-02-27): updates to the pytest tests to fix the issues with pwm, added 2 new tests that bring in 5 total unit tests.  python3 unit tests are passing, this will close #42 and close #47
closed 2017-02-27 by xtacocorex