xtacocorex/CHIP_IO #46: Merge in 2 PWM functions from streamnsight

xtacocorex, 2017-01-05:
Merge in @streamnsight's 2 functions from his fork:
https://github.com/streamnsight/CHIP_IO/commit/51a8663ba4e706305940246cab05f698c3fca631

(Yes, I was totally creeping on the CHIP_IO forks since I noticed the fork number increased)

xtacocorex, 2017-01-28:
@streamnsight I started copying over your 2 PWM functions.  I noticed that in pwm_set_pulse_width_ns() the line to actually set the duty cycle as the pulse width doesn't have a ;.  Is your code complete or just stubbed out?

In looking at pwm_set_period_ns(), that one seems good to go.

xtacocorex, 2017-01-28:
Definitely partially implemented.  I think I got it squared away though.

closed 2017-01-28 by xtacocorex

streamnsight, 2017-01-28:
ah, yeah sorry. i had started but since it didn't work for me, I had put it on hold...

xtacocorex, 2017-01-28:
Well I think I got stuff hashed out, code is pushed to master if you wanted to try it out.  I haven't tested it since my dev CHIP is downstairs and I'm remoted into it working on the code.

commit 73ae207e (2017-01-28): Copied over 2 new pwm functions from @streamnsight to close #46. these 2 functions are untested, which is why i'm not updating the readme at the moment.