xtacocorex/CHIP_IO #48: GPIO.PUD_DOWN

sumarouno, 2017-01-07:
is this lib has GPIO.PUD_DOWN?
Thanks

xtacocorex, 2017-01-07:
The options are there, but they don't actually do anything.  In RPi.GPIO, it sets that setting in the GPIO registers of the Broadcom SOC.  In the BeagleBone Black IO library (from which this is ported), it only set the initial value.  The BBB IO and CHIP_IO don't do direct access to the processor like the the Raspberry Pi library.

I'm pretty sure the code for it got removed in the shuffle when the CHIP moved from a 4.3 kernel to a 4.4 kernel.  I need to look at the prior commits and try to resurrect that code, if it exists.

I'm pretty busy tomorrow with some car maintenance I have to take care of, but I'll change this to a bug to make sure i re-implement whatever I find.

sumarouno, 2017-01-07:
Thank for your reply @xtacocorex . Great explanation

xtacocorex, 2017-01-07:
So in looking at the old code for handling the pull up down stuff, all it did was look to see if it was specified.  If it was set it incorrectly wrote the value to the Pin you were trying to export (in looking I found out why I had some stupid case for just the XIO because I introduced a bug in the code early on in the port).

There is no need for me to actually bring code back in, since what I wanted to bring in is incorrect and doesn't do anything.

I did go look at the BeagleBone Black IO Library just a bit ago and they now use the pull up and down stuff to set the mode on the BBB's OMAP processor.  Their way is specific to that processor, so it won't work on the CHIP.  But their method of doing so involves writing to sysfs, so there is hope that I can find a similar way for the CHIP.

Going to keep this open while I determine what I can actually do to get this actually working.

xtacocorex, 2017-01-29:
Big thanks to @WereCatf for the code help with getting this implemented.  I have verified that the registers are getting set with my code that I'm about to push up that will close this issue.

commit 2b23e2d1 (2017-01-29): pull up/down feature addition to close #48. fixed gpio cleanup() function again, 3rd time is a charm. version bump to 0.3.4
closed 2017-01-29 by xtacocorex