xtacocorex/CHIP_IO #50: GPIO.cleanup() is broken

xtacocorex, 2017-01-09:
I introduced a weird corner case when using GPIO.cleanup() in implementing #43. In the scenario when you just import CHIP_IO.GPIO and immediately do a cleanup on all pins, thing's break.

Test Code:

    import CHIP_IO.GPIO as GPIO
    GPIO.cleanup()

If you actually export a pin from inside CHIP_IO and then do a GPIO.cleanup(), it's happy.

@fordsfords I think the issue is with the C code and the Python API.  I know you know nothing about the Python API part, but could you look at: https://github.com/xtacocorex/CHIP_IO/commit/8ecec67bad0f221135faaaec25463a64880d9404
(py_cleanup in py_gpio.c) and see if I'm doing that strcmp right? 

My initial debugging had me put an if statement around the close(epfd) in event_cleanup() inside event_gpio.c.  That didn't help the situation.

xtacocorex, 2017-01-10:
Disregard @fordsfords, I just figured it out with the help of gdb. Woo!

closed 2017-01-10 by xtacocorex

howientc, 2017-01-19:
Hi,

I think there's still a bug here. From a .py file, importing CHIP_IO and then calling GPIO.cleanup() throws an exception. Weirdly, the same from a python shell does not on my CHIP.  I poked around in your commit log a bit, and figured out that calling GPIO.cleanup("") (with an empty string) avoids the exception.

xtacocorex, 2017-01-19:
Howie, was this on a fresh checkout?  I'm nowhere near my CHIPs at the moment, left the PocketCHIP at home.  I know I was getting an error last week and I think I have code on my dev CHIP at home that I haven't checked in yet.  I'll let you know when I get back home and get squared away.

howientc, 2017-01-19:
Yes.  Fresh.  Calling with empty string avoids exception, but it doesn't
look like it's working.  I would expect cleanup() to unexport all pins,
 but it doesn't do that.  I am calling cleanup() at start of my code. My
code raises exceptions that pins are already exported when I try to use
them after the cleanup() call.

Howie

xtacocorex, 2017-01-19:
So, the cleanup will only unexport pins that it knows are exported. The addition of pin specification was added for my REST API for the CHIP.

If you need to blast away all pins at the beginning of your script, look at unexport_all() in CHIP_IO.Utilities.

I'll look into this when I get home and situated, currently in Detroit about to board a plane to Iowa.

howientc, 2017-01-20:
Thanks. Calling the script works for my purposes.

xtacocorex, 2017-01-24:
Howie, interestingly enough, my setup on my Dev CHIP was 2 commits behind master.  So I updated it.  In my initial testing, I don't get any errors/exceptions:

    [chip@chipdev][02:32][test]: sudo python
    Python 2.7.9 (default, Aug 13 2016, 17:56:53)
    [GCC 4.9.2] on linux2
    Type "help", "copyright", "credits" or "license" for more information.
    >>> import CHIP_IO.GPIO as gp
    >>> gp.cleanup()
    >>> gp.setup("XIO-P3",gp.IN)
    >>> gp.cleanup()
    >>> gp.setup("XIO-P3",gp.IN)
    >>> gp.cleanup("XIO-P3")
    >>>

commit 486cf738 (2017-01-10): Fix and close #50, I really don't understand pointers because Python abstracts that all out for me and C does not.