xtacocorex/CHIP_IO #51: Make Issue Python 3

tryonlinux, 2017-01-11:
New to GitHub, so I apologize if this is the wrong place for this, but I was having issues compiling when getting to the below step:

sudo python3 setup.py instal
 
The issue was something about rpigpiomodule not being defined (forgot to copy it) in py_gpio.c. So I looked at the adafruit-beagalbone repository and saw how they defined it there (https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/py_gpio.c) and added it to my local copy of py_gpio.c  at line 862 (Code I added is below). It now compiles without issues. When I figure out how to commit changes back to the repository I will. Everything works perfect now. Thanks for your hard work porting this!! 

```
#if PY_MAJOR_VERSION > 2
static struct PyModuleDef rpigpiomodule = {
   PyModuleDef_HEAD_INIT,
   "GPIO",       // name of module
   moduledocstring,  // module documentation, may be NULL
   -1,               // size of per-interpreter state of the module, or -1 if the module keeps state in global variables.
   gpio_methods
};
#endif
```

tryonlinux, 2017-01-11:
Actually I see that you can just replace line 864 

`   if ((module = PyModule_Create(&rpigpiomodule)) == NULL)
`

with

`   if ((module = PyModule_Create(&chipgpiomodule)) == NULL) 
`

Didn't even notice that...my bad

xtacocorex, 2017-01-11:
I wondered if that fix would break something. Damn.  WIll fix asap.

tryonlinux, 2017-01-11:
Thank you!  I forked it and recompiled and the use of the gpio seems to be working for me.

xtacocorex, 2017-01-11:
I swore I had changed those 2 lines a year ago when I ported the library, how it got reverted is beyond me.  Fixed. :)

closed 2017-01-11 by xtacocorex

tryonlinux, 2017-01-11:
Awesome thank you,  and thanks a ton for porting it!

commit 10e70432 (2017-01-11): Fix and close #51, I wasn't careful in swapping the module name. I thought the name was changed months ago, but eh, fixed now