xtacocorex/CHIP_IO #54: malloc without a free

Flexshot, 2017-01-12:
In pwm_start we malloc memory for pwm struct in pwm_start but don't free it.

https://github.com/xtacocorex/CHIP_IO/blob/master/source/c_pwm.c#L367

Flexshot, 2017-01-12:
I think it gets free here. https://github.com/Flexshot/CHIP_IO/blob/master/source/c_pwm.c#L451 
Let's verify in all cases it's getting freed.

xtacocorex, 2017-01-13:
This is original code from the BBB.  @fordsfords, you are well versed in the language of C and I know literally nothing about malloc and free commands.  Do you see anything wrong with how the code is setup?  Does the memory actually get freed up to the system properly?

fordsfords, 2017-01-13:
I'm traveling at the moment and can't easily do any testing.  But @Flexshot is correct.  Basically there is a global variable which is the head of a linked list of these pwm structures.  As best as I can see via code inspection, the functions which add and delete pwms from that global list are correctly written.  So the only risk I know of for memory leaks would be a Python program allocating a pwm and then itself not disposing of it.  That would be a bug in the user's program, not in the CHIP_IO library.

But I'll also mention that the code is not thread-safe.  Which is to say, multiple threads asynchronously creating and/or deleting pwms from that linked list are likely to collide in fatal ways.

However, this might not be an actual problem.  I know that Python supports multiple threads, but it does so in an unusual way.  I don't remember the details, but it would not surprise me if these C extensions are considered "atomic" from a threading perspective.  I.e. there should never be two threads executing the same C code at the same time.  But unfortunately, I'm not very confident in my Python threading knowledge, so this continues to be a possible area of danger.  (For what it's worth, this is not the only non-thread safe code I've seen in CHIP_IO.)

xtacocorex, 2017-01-29:
With the debug turned on while using the interactive Python interpreter, I noticed that GPIO cleanup gets called when Ctrl+D is pressed to exit out of the interpreter.

I believe the memory is freed automatically on script exit.

As far as threading goes, I don't know what would need to be done to make it thread safe.  

I'm going to close this as it appears things are good to go with the memory.

closed 2017-01-29 by xtacocorex