xtacocorex/CHIP_IO #57: Error setting up channel X

nio101, 2017-01-29:
Hello,

Once a python script using CHIP_IO has exited, when I try to run it again, even with using `GPIO.cleanup()` before setting up the IO, I'm getting an error:
```
GPIO.setup(set, GPIO.OUT)
RuntimeError: Error setting up channel CSID2, maybe already exported? (gpio_export: could not write '134' to /sys/class/gpio/export (Device or resource busy))
```
Is there any workaround to be able to run my script again, without rebooting?
Thanks.

xtacocorex, 2017-01-29:
So, it looks like your script isn't unexporting CSID2, so when you start it again, it's throwing the error.

What does your full script look like?  I think you might not be cleaning up the GPIO properly.

There is a way to deal with this, the Utilities module.  You can do the following at the beginning of your script:

    import CHIP_IO.Utilities as UT
    UT.unexport_all()

The above will work, but it doesn't solve the error of the GPIO not being cleaned up.

nio101, 2017-01-29:
Thank you, I will try this, and also handle correctly the signal supervisord is sending to my script to gracefully shut it down... :)

xtacocorex, 2017-02-03:
@nio101, you still have issues with cleanup?  My latest commits here fix it for good now (hopefully).  Just seeing if there is an issue still, otherwise I'll close this.

closed 2017-02-05 by xtacocorex