xtacocorex/CHIP_IO #58: Most recent update breaks setting up XIO pins as input

brettcvz, 2017-01-29:
Run `GPIO.setup("XIO-P0", GPIO.IN)`, get:
```
ValueError: Pull Up/Down setting not capable for XIO-P0. ()
```
The problem is at https://github.com/xtacocorex/CHIP_IO/commit/2b23e2d1654b0febaef9c7d5ac61feb5f2da823a#diff-6021bb287e36c7d509f9435891dba0beR190

I was really surprised by this because my code was working last night and then broke just now when I re-cloned and re-built this library. I'll see if I can get a fix in soon, but wanted to flag this ASAP.

xtacocorex, 2017-01-29:
I did some testing to make sure this wouldn't happen.  I know how to fix it.

brettcvz, 2017-01-29:
👍 thanks for the fast response. I think just wrapping it in `if` clause to skip the code if pud=PUD_OFF should do the trick.

xtacocorex, 2017-01-29:
It's an error in the logic, I shouldn't throw an exception if the pin can't handle PUD setting.  Luckily I was working on adding PWM and SoftPWM to my CHIP REST API, so I have all the pages open.

xtacocorex, 2017-01-29:
Should be good to go now! I also pushed to pypi if you installed it via pip.

closed 2017-01-29 by xtacocorex

brettcvz, 2017-01-29:
Thanks! Works great.

commit 2e2177da (2017-01-29): quick fix because i broke xio input setup with the pud code, this closes #58