xtacocorex/CHIP_IO #60: I was not able to install it on the C.H.I.P

shaddow501, 2017-02-14:
Dear Sir

I am getting this error:

/usr/include/features.h:374:25: fatal error: sys/cdefs.h: No such file or directory
 #  include <sys/cdefs.h>

xtacocorex, 2017-02-14:
I need more information than just this.  

What file caused this error?  
Did you install all the proper dependencies per the README?
Can you provide the full build output?

shaddow501, 2017-02-14:
Dear Robert

 

Tell me what to do and I will send you everything you need, I am not Linux expert…

 

I did the following:

 

sudo apt-get update

sudo apt-get install git build-essential python-dev python-pip flex bison chip-dt-overlays -y

git clone https://github.com/atenart/dtc.git

cd dtc

make

sudo  make install PREFIX=/usr

cd ..

git clone git://github.com/xtacocorex/CHIP_IO.git

cd CHIP_IO

sudo python setup.py install

cd ..

and stopped with make error as  I have sent to you.

shaddow501, 2017-02-14:
root@chip:/home/chip/CHIP_IO/dtc# ls

checks.c                   dtc-parser.y  Makefile.convert-dtsv0

checks.d                   dtdiff        Makefile.dtc

convert-dtsv0-lexer.l      fdtdump.c     Makefile.utils

convert-dtsv0-lexer.lex.c  fdtdump.d     README

convert-dtsv0-lexer.lex.d  fdtget.c      README.license

data.c                     fdtget.d      scripts

data.d                     fdtput.c      srcpos.c

Documentation              fdtput.d      srcpos.d

dtc.c                      flattree.c    srcpos.h

dtc.d                      flattree.d    tests

dtc.h                      fstree.c      TODO

dtc-lexer.l                fstree.d      treesource.c

dtc-lexer.lex.c            GPL           treesource.d

dtc-lexer.lex.d            libfdt        util.c

dtc-parser.tab.c           livetree.c    util.d

dtc-parser.tab.d           livetree.d    util.h

dtc-parser.tab.h           Makefile      version_gen.h

 

 

root@chip:/home/chip/CHIP_IO/dtc# make

        CHK version_gen.h

         CC srcpos.o

In file included from /usr/include/stdio.h:27:0,

                 from srcpos.c:22:

/usr/include/features.h:374:25: fatal error: sys/cdefs.h: No such file or directory

#  include <sys/cdefs.h>

                         ^

compilation terminated.

Makefile:255: recipe for target 'srcpos.o' failed

make: *** [srcpos.o] Error 1

root@chip:/home/chip/CHIP_IO/dtc#

 

Next:

 

root@chip:/home/chip/CHIP_IO# sudo python setup.py install

Compiling DTS Files

Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property

Checking for DTBO Install Path

Removing old DTBO files (if applicable)

Moving DTBO files to /lib/firmware/chip_io

running install

running bdist_egg

running egg_info

writing CHIP_IO.egg-info/PKG-INFO

writing top-level names to CHIP_IO.egg-info/top_level.txt

writing dependency_links to CHIP_IO.egg-info/dependency_links.txt

reading manifest file 'CHIP_IO.egg-info/SOURCES.txt'

reading manifest template 'MANIFEST.in'

writing manifest file 'CHIP_IO.egg-info/SOURCES.txt'

installing library code to build/bdist.linux-armv7l/egg

running install_lib

running build_py

running build_ext

building 'CHIP_IO.GPIO' extension

arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c source/py_gpio.c -o build/temp.linux-armv7l-2.7/source/py_gpio.o -Wno-format-security

In file included from /usr/include/stdlib.h:24:0,

                 from source/py_gpio.c:39:

/usr/include/features.h:374:25: fatal error: sys/cdefs.h: No such file or directory

#  include <sys/cdefs.h>

                         ^

compilation terminated.

error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

root@chip:/home/chip/CHIP_IO#

xtacocorex, 2017-02-15:
Hmm, it appears libc6 might not have gotten installed.

Try:

    sudo apt-get install libc6 libc6-dev

And then rebuild dtc and CHIP_IO again.

shaddow501, 2017-02-15:
I think I did before install them

libc6 is already the newest version.

libc6-dev is already the newest version.

 

Finally after looking that file I was able to find 3 packages that were missing and apt-get install said that there is no release candidate or they weren’t found.

 

I managed to compile it all downloading the files manually move them to the CHIP home folder and installed them:

 

 

https://packages.debian.org/sid/libc6-dev-armhf-cross

 

dpkg –I libc6-dev-armhf-cross_2.24-9cross1_all.deb

dpkg –I linux-libc-dev-armhf-cross_4.9.2-2cross1_all.deb

dpkg –I libc6-armhf-cross_2.24-9cross1_all.deb 

 

Sorry not sure if in this order…

xtacocorex, 2017-02-15:
So CHIP_IO is now happy for you?

I did notice you're running as root.  That is not recommended on Linux machines as you can destroy your system if you don't know what you're doing.

Are you on a CHIP or CHIP Pro?

shaddow501, 2017-02-15:
Lol Not working yet

Now I have other problem…

 

I am on CHIP not CHIP Pro

 

About being root, I do many backups :)

 

I made this test file:

 

import CHIP_IO.GPIO as GPIO

import time

GPIO.cleanup()

GPIO.setup("XIO-P2", GPIO.OUT)

print "Toggling XIO-P1 10 times..."

for i in range(0,10):

GPIO.output("XIO-P2", GPIO.LOW)

time.sleep(0.1)

GPIO.output("XIO-P2", GPIO.HIGH)

time.sleep(0.1)

GPIO.cleanup()

 

When I run it I get:

root@chip:/home/chip/CHIP_IO# python led.py 

  File "led.py", line 13

    GPIO.output("XIO-P2", GPIO.LOW)

       ^

IndentationError: expected an indented block

 

How to fix it now?

xtacocorex, 2017-02-15:
Python is very strict about indentation.  The indents need to be the same number of spaces.  The code below is indented properly below.

    import CHIP_IO.GPIO as GPIO
    import time

    GPIO.cleanup()
    GPIO.setup("XIO-P2", GPIO.OUT)
    print "Toggling XIO-P1 10 times..."

    for i in range(0,10):
        GPIO.output("XIO-P2", GPIO.LOW)
        time.sleep(0.1)
        GPIO.output("XIO-P2", GPIO.HIGH)
        time.sleep(0.1)

    GPIO.cleanup()

shaddow501, 2017-02-15:
Sorry

 

My mistake

 

I copied it wrong…

 

import CHIP_IO.GPIO as GPIO

import time

GPIO.cleanup()

GPIO.setup("XIO-P2", GPIO.OUT)

 

print "Toggling  XIO-P1 10 times..."

for i in range(0,10):

        GPIO.output("XIO-P2", GPIO.LOW)

        time.sleep(0.1)

        GPIO.output("XIO-P2", GPIO.HIGH)

        time.sleep(0.1)

GPIO.cleanup()

shaddow501, 2017-02-15:
Yes sorry to bother you figured it out later lol

 

It is almost 5 am and I was trying to figure out what I do wrong with the CHP-IO for about 2 days, when I almost lost hope I did contact you.

 

It might that others will encounter the same issue as I did so if you want add those 3 packages that I had to install in not apt-get install way…

 

Thanks for your help

xtacocorex, 2017-02-15:
No worries!  I wonder if you did something to your system that caused those to not be there.  I haven't had to install them on any of my CHIPs and my CHIP Pro to get CHIP_IO to install.

I'll keep this open and investigate a bit.

shaddow501, 2017-02-15:
I am on the latest kernel 4.4 something.

 

I just flashed it with the getchip no limit version and uploaded the tar –zxvpf.

If the files weren’t deleted or anything just wasn’t there..

shaddow501, 2017-02-15:
This is the backup I do

 

For backup:

cd /

mkdir /backup

mount /dev/sda1 /backup

tar -zcvpf /backup/fullbackup.tar.gz --directory=/ --exclude=proc --exclude=sys --exclude=dev \

--exclude=sys --exclude=mnt --exclude=backup --exclude='./etc/udev/rules.d/70-persistent-net.rules' .

shaddow501, 2017-02-15:
Now I have this problem

Also doesn’t work after reboot…

 

    GPIO.setup("LCD-D23", GPIO.OUT)

RuntimeError: Error setting up channel LCD-D23, maybe already exported? (gpio_export: could not write '119' to /sys/class/gpio/export (Device or resource busy))

 

root@chip:/sys/class/gpio# ls

export  gpiochip0  gpiochip1013  gpiochip1021  unexport

 

I cannot unexport them…

xtacocorex, 2017-02-15:
I'm beginning to think your backup might be causing issues, although I'm not 100% sure on that.
How often are you backing up and restoring your CHIP?

I have no issues with LCD-D23 on the CHIP I use for development.  

    [chip@chipdev][04:13][~]: sudo python
    [sudo] password for chip:
    Python 2.7.9 (default, Aug 13 2016, 17:56:53)
    [GCC 4.9.2] on linux2
    Type "help", "copyright", "credits" or "license" for more information.
    >>> import CHIP_IO.GPIO as gp
    >>> gp.setup("LCD-D23",gp.OUT)
    >>> gp.cleanup()
    >>> gp.toggle_debug()
    >>> gp.setup("LCD-D23",gp.OUT)
     ** gpio_export **
     ** gpio_export: creating data struct **
     ** gpio_set_direction: out **
     ** gpio_set_pud: port 3, pin 23, value 0 **
     ** gpio_get_pud: port 3, pin 23 **
     ** open_value_file **
    >>> gp.cleanup()
     ** exports_cleanup **
     ** gpio_unexport **
     ** gpio_unexport: freeing memory **
     ** gpio_unexport **
    >>>

shaddow501, 2017-02-15:
I backup after a major change.
I didnt backup yet after getting the CHIP IO working.

The problem happen I think is when I stopped python script during process....
But now I cannot remove any of the gpiochip1013 and 1021...

Ilan B. (Mobile)

https://github.com/xtacocorex/CHIP_IO/issues/60#issuecomment-279913633

shaddow501, 2017-02-15:
Update

 

It seems that after adding this line to the python script it fixed the problem…

 

Before that I did run Utilities.py (python Utilities.py)

 

Could that fixed my problem?

 

def unexport_all():

    if DEBUG:

        print("Unexporting all the pins")

    gpios = glob.glob("/sys/class/gpio/gpio[0-9]*")

    for g in gpios:

        tmp = g.split("/")

        gpio = tmp[4]

        num = re.sub("[a-z]","",gpio)

        cmd = "echo " + num + " > /sys/class/gpio/unexport"

        subprocess.Popen(cmd,shell=True, stdout=subprocess.PIPE)

 

 

There is a command to free al GPIOs but it doesn’t work…

 

root@chip:/home/chip/CHIP_IO# sh unexport_all.sh 

/sys/class/gpio/gpio[0-9]* 0-9]*

unexport_all.sh: line 6: echo: write error: Invalid argument

 

I don’t know why, I am not an expert but it seems that with this configuration python script everything works well with no issues even when I stop (CTRL-C) the script it still working after with no issues.

 

In the script I toggle between 2 x GPIOs  as you can see.

One leg of the LED is connected to D23 and the other to D21 and I just switch between them.

It is just for testing purpose I use 2 GPIOs to drive a LED, I know that I can use only one ((-) leg of the LED to the GND, or (+) leg to the VCC, but just wanted to test it.

 

import CHIP_IO.GPIO as GPIO

import time

import subprocess

import glob

import re

print "First freeing GPIOs"

gpios = glob.glob("/sys/class/gpio/gpio[0-9]*")

for g in gpios:

        tmp = g.split("/")

        gpio = tmp[4]

        num = re.sub("[a-z]","",gpio)

        cmd = "echo " + num + " > /sys/class/gpio/unexport"

        subprocess.Popen(cmd,shell=True, stdout=subprocess.PIPE)

GPIO.cleanup()

#GPIO.toggle_debug()

print "Now allocating GPIO"

 

GPIO.setup("LCD-D23", GPIO.OUT)

GPIO.setup("LCD-D21", GPIO.OUT)

print "Toggling  LCD-D21 & LCD-D23 10 times..."

for i in range(0,1000):

        GPIO.output("LCD-D21", GPIO.LOW)

        GPIO.output("LCD-D23", GPIO.HIGH)

        time.sleep(0.05)

        GPIO.output("LCD-D21", GPIO.HIGH)

        GPIO.output("LCD-D23", GPIO.LOW)

        time.sleep(0.05)

        GPIO.output("LCD-D21", GPIO.LOW)

        GPIO.output("LCD-D23", GPIO.LOW)

GPIO.cleanup()

xtacocorex, 2017-02-15:
You probably didn't import the CHIP_IO Utilities correctly.  You don't need an unexport_all() function as I have one.  Here is what your code should be like:

    import CHIP_IO.GPIO as GPIO
    import CHIP_IO.Utilities as UT
    import time
    import subprocess
    
    print "First freeing GPIOs"
    # This will blast away all the GPIO exported on the system
    # use wisely if other scripts are running that are accessing
    # GPIO
    UT.unexport_all()
    
    # No need to do a GPIO.cleanup here as nothing
    # has been exported, so the underling module
    # code won't do anything
    
    #GPIO.toggle_debug()
    
    print "Now allocating GPIO"
    GPIO.setup("LCD-D23", GPIO.OUT)
    GPIO.setup("LCD-D21", GPIO.OUT)
    
    print "Toggling  LCD-D21 & LCD-D23 10 times..."
    for i in range(0,1000):
        GPIO.output("LCD-D21", GPIO.LOW)
        GPIO.output("LCD-D23", GPIO.HIGH)
        time.sleep(0.05)
        GPIO.output("LCD-D21", GPIO.HIGH)
        GPIO.output("LCD-D23", GPIO.LOW)
        time.sleep(0.05)
        GPIO.output("LCD-D21", GPIO.LOW)
        GPIO.output("LCD-D23", GPIO.LOW)
    
    # Final Cleanup
    GPIO.cleanup()

shaddow501, 2017-02-15:
Yes it looks much better than what I did

 

Thank you

 

What if I would like to unexport only the GPIOs that I have used in the script?

Lets say that there are other python processes that might use other GPIOs and if I set UT.unexport_all() it will damage them as well.
 
Is the right command UT.unexport_All("LCD-D23","LCD-D21")   ?

xtacocorex, 2017-02-15:
It is not possible to provide arguments to UT.unexport_all().  It's mainly for development so that if the script screws up, you can cleanly enter when you run the script again.  

Your earlier problem was probably the Ctrl-C.  The code isn't handling the KeyboardInterrupt exception that is thrown, so the GPIO are still exported when the script is force closed.

That can be solved by changing the code to:

    import CHIP_IO.GPIO as GPIO
    import CHIP_IO.Utilities as UT
    import time
    import subprocess
    
    print "First freeing GPIOs"
    # This will blast away all the GPIO exported on the system
    # use wisely if other scripts are running that are accessing
    # GPIO
    # Recommended to not use in Live systems
    #UT.unexport_all()
    
    # No need to do a GPIO.cleanup here as nothing
    # has been exported, so the underling module
    # code won't do anything
    
    #GPIO.toggle_debug()
    
    print "Now allocating GPIO"
    GPIO.setup("LCD-D23", GPIO.OUT)
    GPIO.setup("LCD-D21", GPIO.OUT)
    
    print "Toggling  LCD-D21 & LCD-D23 10 times..."
    # The try/catch will grab the Ctrl-C from the user 
    # while the script is running
    try:
        for i in range(0,1000):
            GPIO.output("LCD-D21", GPIO.LOW)
            GPIO.output("LCD-D23", GPIO.HIGH)
            time.sleep(0.05)
            GPIO.output("LCD-D21", GPIO.HIGH)
            GPIO.output("LCD-D23", GPIO.LOW)
            time.sleep(0.05)
            GPIO.output("LCD-D21", GPIO.LOW)
            GPIO.output("LCD-D23", GPIO.LOW)
    catch KeyboardInterrupt:
        # This will hit anytime Ctrl-C is pressed
        GPIO.cleanup()
        
    # Final Cleanup
    # This will get hit if the loop runs through 
    # all it's iterations
    GPIO.cleanup()

shaddow501, 2017-02-15:
Thanks a lot

Ilan B. (Mobile)

    
    
    
    
    
        

https://github.com/xtacocorex/CHIP_IO/issues/60#issuecomment-280021279

shaddow501, 2017-02-15:
Deat Sir 

 

Your option didn’t work for me it, the python reported errors.

I finally did that instead and I think it is working.

 

try:

     for i in range(0,1000000):

        GPIO.output("LCD-D21", GPIO.LOW)

        GPIO.output("LCD-D23", GPIO.HIGH)

        time.sleep(0.05)

        GPIO.output("LCD-D21", GPIO.HIGH)

        GPIO.output("LCD-D23", GPIO.LOW)

        time.sleep(0.05)

        GPIO.output("LCD-D21", GPIO.LOW)

        GPIO.output("LCD-D23", GPIO.LOW)

     GPIO.cleanup()

except KeyboardInterrupt:

       GPIO.cleanup()

 

Lol after I have finally learned how to program correctly in Arduino IDE, now python is a new thing for me…

xtacocorex, 2017-02-15:
I am at work at the moment and created that script above on the fly without testing.  

You should read up on how to code in Python; the issue I caused in the sample above is in due to where I placed the exception handling.  It is not caused by CHIP_IO.

shaddow501, 2017-02-15:
It is ok all works now

closed 2017-02-19 by xtacocorex

adiman73, 2018-11-12:
Hello.
You can help me install the "https://github.com/atenart/dtc" libraries as they are no longer available on that link.     
This tutorial    https://drive.google.com/open?id=1DhiZDBd-VJB6IrRimrW3xskTsH8QevHW    . Thanks﻿