xtacocorex/CHIP_IO #61: Polarity not set for normal CHIP

pschichtel, 2017-03-05:
I spent quite some time figuring out why my code didn't use my polarity setting just to find this line:

https://github.com/xtacocorex/CHIP_IO/blob/master/source/c_pwm.c#L531

Is there still a reason for the CHIP-PRO check? Setting it manually using ```echo "normal" > /sys/class/pwm/pwmchip0/pwm0/polarity``` worked just fine.

xtacocorex, 2017-03-05:
Hmm.  How are you loading the overlay?  I had issues early on when I was fixing the PWM code and polarity never worked right.  I can definitely revisit this.

xtacocorex, 2017-03-05:
Here are my quick tests on a normal CHIP:

    root@chipdev:/sys/devices/platform/soc@01c00000/1c20e00.pwm/pwm/pwmchip0/pwm0# cat enable
    0
    root@chipdev:/sys/devices/platform/soc@01c00000/1c20e00.pwm/pwm/pwmchip0/pwm0# echo normal > polarity
    root@chipdev:/sys/devices/platform/soc@01c00000/1c20e00.pwm/pwm/pwmchip0/pwm0# echo inverted > polarity
    bash: echo: write error: Invalid argument
    root@chipdev:/sys/devices/platform/soc@01c00000/1c20e00.pwm/pwm/pwmchip0/pwm0# echo 1 > enable
    root@chipdev:/sys/devices/platform/soc@01c00000/1c20e00.pwm/pwm/pwmchip0/pwm0# echo normal > polarity
    bash: echo: write error: Device or resource busy
    root@chipdev:/sys/devices/platform/soc@01c00000/1c20e00.pwm/pwm/pwmchip0/pwm0# echo inverted > polarity
    bash: echo: write error: Invalid argument
    root@chipdev:/sys/devices/platform/soc@01c00000/1c20e00.pwm/pwm/pwmchip0/pwm0# echo 0 > enable

I am remembering now that PWM has to have enable be 0 in order to change the polarity.

So I ssh into my CHIP Pro DevKit since it has PWM built into the main dtb on boot.  Interestingly enough, when you cat the polarity, you get this: 

    root@chippy:/sys/devices/platform/soc@01c00000/1c20e00.pwm/pwm/pwmchip0/pwm0# cat polarity
    inversed

The CHIP_IO code tries to write inverted, not inversed.

Literally a one letter bug.

closed 2017-03-05 by xtacocorex

pschichtel, 2017-03-05:
thanks for the quick fix, I did that typo as well while using the shell... 😆

xtacocorex, 2017-03-06:
Hey, no worries!  I didn't think that I was literally off by a single letter when I was working on the code months ago trying to fix PWM (there used to be many issues).  Thanks for bringing this to my attention.

At least I wasn't the only one to make the mistake. :)

commit dcd8ea6f (2017-03-05): Re-enabled polarity set on the CHIP PWM, fixed issues that caused me to disable that feature to begin with. This could have been a 1 letter change, but I re-eabled tests in the unit test to validate. This will close #61