xtacocorex/CHIP_IO #62: adding callback functions on any pin that supports edge detection not work

aleksandr-rakov, 2017-03-06:
this example from README
```
(env)root@chip:~/src/readme# cat example2.py 
import CHIP_IO.GPIO as GPIO

def mycallback(channel):
    print("we hit the edge we want")

GPIO.setup("GPIO3", GPIO.IN)
# Add Callback: Falling Edge
GPIO.add_event_callback("GPIO3", GPIO.FALLING, mycallback)
# Add Callback: Rising Edge
GPIO.add_event_callback("GPIO3", GPIO.RISING, mycallback)
# Add Callback: Both Edges
GPIO.add_event_callback("GPIO3", GPIO.BOTH, mycallback)
# Remove callback
GPIO.remove_event_detect("GPIO3")
```

```
(env)root@chip:~/src/readme# python example2.py 
Segmentation fault
```

```
(env)root@chip:~/src/readme# uname -a
Linux chip 4.4.13-ntc-mlc #1 SMP Tue Dec 6 21:38:00 UTC 2016 armv7l GNU/Linux
(env)root@chip:~/src/readme# cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 8 (jessie)"
NAME="Debian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=debian
HOME_URL="http://www.debian.org/"
SUPPORT_URL="http://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
  BUILD_ID=Fri Dec  9 22:28:28 UTC 2016
  VARIANT="Debian on C.H.I.P"
  VARIANT_ID=DESKTOP
```

xtacocorex, 2017-03-06:
Thanks for submitting the issue, it's definitely legit and I'll start taking a look at it.  I believe it's due to my swapping things around for the pull up/down resistor settings for the R8 based GPIO.

Just a note, the README examples are just there to show the API; you can only have one callback per Pin.  I should make a note of that in the README.

xtacocorex, 2017-03-06:
Thanks again!  This ended up being an missed update after copy/pasting similar code to handle the Pull Up/Down settings.

closed 2017-03-06 by xtacocorex

xtacocorex, 2017-03-06:
And crap, forgot to update the README. :(

Just know that trying to do more than one callback is not supported and an error will be thrown.

xtacocorex, 2017-03-07:
And pushed a fix to the README to update the edge detection section.

aleksandr-rakov, 2017-03-07:
GPIO.wait_for_edge and GPIO.add_event_detect works fine

but GPIO.add_event_callback still not work
now error is TypeError: an integer is required

```
(env)root@chip:~/src/readme2# cat example3.py 
import CHIP_IO.GPIO as GPIO

def mycallback(channel):
    print("we hit the edge we want")

GPIO.setup("GPIO3", GPIO.IN)
GPIO.add_event_callback("GPIO3", GPIO.FALLING, mycallback)
GPIO.remove_event_detect("GPIO3")
```
```
(env)root@chip:~/src/readme2# python example3.py 
Traceback (most recent call last):
  File "example3.py", line 7, in <module>
    GPIO.add_event_callback("GPIO3", GPIO.FALLING, mycallback)
TypeError: an integer is required
```

```
(env)root@chip:~/src/readme2# cat example4.py 
import CHIP_IO.GPIO as GPIO

def mycallback(channel):
    print("we hit the edge we want")

GPIO.setup("XIO-P0", GPIO.IN)
GPIO.add_event_callback("XIO-P0", GPIO.FALLING, mycallback)
GPIO.remove_event_detect("XIO-P0")
```

```
(env)root@chip:~/src/readme2# python example4.py 
Traceback (most recent call last):
  File "example4.py", line 7, in <module>
    GPIO.add_event_callback("XIO-P0", GPIO.FALLING, mycallback)
TypeError: an integer is required
```

xtacocorex, 2017-03-07:
The issue is with the README being incorrect, yet again.

add_event_callback() does not handle the edge as an input.

You need to do the following:

    GPIO.add_event_detect("GPIO3", GPIO.FALLING)
    GPIO.add_event_callback("GPIO3", mycallback)

or

    GPIO.add_event_detect("GPIO3", GPIO.FALLING, mycallback)

I'm updating the README yet again.

Also, your example code (based on the README) will only set the callback and then immediately remove it.  The event won't be triggered.  You are also not cleaning up the GPIO, so there will be issues when re-running due to the pin being exported.

xtacocorex, 2017-03-07:
Just pushed a fix to the README.

aleksandr-rakov, 2017-03-07:
Many thanks! Now everything works fine)

commit 4da0812a (2017-03-06): Fixing issue where using alt name for the XIO would cause a segfault when speficying the pin as an input.  This will close #62