xtacocorex/CHIP_IO #63: setup doesn't work twice

danjperron, 2017-03-19:
I got a problem when I want to use your module with my burnLVPx.py . PIC programmer.

Since I'm switching the pin from INPUT to OUTPUT all the times  the program stop and display an error about the pin is already exported.

To fix the problem I had to add in event_gpio.c  function gpio_export()  a check if the pin is already exported. if it is then unexport it first.

int gpio_export(int gpio)
{
    int fd, len, e_no;
    char filename[MAX_FILENAME];
    char str_gpio[80];
    struct gpio_exp *new_gpio, *g;

    if (DEBUG)
        printf(" ** gpio_export **\n");

   // verify if it is already exported
   // if it is ! unexport first
   // djp  March 18,2017
   snprintf(filename, sizeof(filename), "/sys/class/gpio/gpio%d", gpio);BUF2SMALL(filename);
   fd = open(filename,O_RDONLY);
   if(fd >=0)
   {
     close(fd);
     gpio_unexport(gpio);
   }

    snprintf(filename, sizeof(filename), "/sys/class/gpio/export"); BUF2SMALL(filename);

xtacocorex, 2017-03-19:
Dan,

Thanks for the heads up on this.  Turns out, this feature exists already as someone else was trying to do exactly what you are doing, so they submitted a pull request that I merged in a while back.  The feature just missed getting added to the README. :(

GPIO.direction(channel, direction) is what you want.  I haven't actually used this feature in any of my code, but you have to setup the pin first, then you're allowed to run this function to toggle the pin direction as needed. 

I'll keep this open so I can add it to the README.

danjperron, 2017-03-19:
Then I will have to make my code different from the Raspberry Pi for  the C.H.I.P. 
I don't thing  that GPIO.direction exist on RPi.GPIO.

Thanks anyway I will modify my code to add the direction fonction

Good job anyway,
Daniel Perron

xtacocorex, 2017-03-19:
Ah, totally understand.

I'll see about getting this added into the setup function without having to tear it down and bring it back up, that operation takes too long.  I powered down my dev chip earlier since it was sitting on my daughters sewing table, so I'll get it set back up here sometime and dive into this.  

Will let you know when I get this done, might be a couple of days.

danjperron, 2017-03-19:
Thanks, 

  In the mean time I will modify my code and add a INOUT function which will select the correct function if it is a C.H.I.P. or a RPI. 

Then I won't need that modification.

It will be done in 30 mins.

Best regards,

Daniel

danjperron, 2017-03-19:
Ok I fixed my code using a overriding class to take care of the differences between the C.H.I.P. and the RPi.
Hi,
During me sleep I thought about it and in reality why an existing export is an error ? It should be a warning and should be able to disable the warning. I know this mean a lot of change in the coding.

Also I did put a patch on the export command but it should have been into  the setup(). If it is already exported than use direction() instead.

I hope that I don't bother you too much about that.

Thank you,
 keep up the good work,
Daniel

xtacocorex, 2017-03-19:
You're definitely not being a bother Daniel.  I was thinking about it too, after looking at your code.  You're handling the difference between RPi.GPIO and CHIP_IO correctly, which is how things should really be done.

I think I have a good idea on how to handle this, just got to get time to implement.

commit 2b48571e (2017-03-21): Updating setup to allow for repeated calls on a pin.  Not totally sold on this, but it work and is what RPi.GPIO does.  This will close #63
closed 2017-03-21 by xtacocorex