xtacocorex/CHIP_IO #69: Edge/Interrupt detection on the C.H.I.P. Pro

jdinard, 2017-05-04:
The current library doesn't support edge detection for any C.H.I.P. Pro exposed pins. The XIO pins are not found on the PRO, and APEINT1 and 3 are not exposed on the PRO.

From the documentation, it seems like we might be able to get this working on EINT13 (Pin 10) to give the PRO interrupt detection. Its listed as PWM1, and EINT13, but is also documented as an external interrupt pin.

I'm far from a great linux programmer, and I would really appreciate help banging this together.

jdinard, 2017-05-04:
This seems to work quite well. All I needed to do was add the checks in stop py_gpio.c to stop the library from rejecting a callback being added to the pin.

xtacocorex, 2017-05-05:
What version of the library are you using?  v0.5.8 added the ability to set PWM1, I2S-MCLK, and I2S-DI as edge detectable pins.  I didn't hook up the 2 UART pins as they are in use by uboot.

Sample code from the last commit.
`
   // check to ensure gpio is one of the allowed pins
   if (gpio != lookup_gpio_by_name("AP-EINT3")
       && gpio != lookup_gpio_by_name("AP-EINT1")
       && gpio != lookup_gpio_by_name("I2S-MCLK") // CHIP PRO
       && gpio != lookup_gpio_by_name("I2S-DI") // CHIP PRO
       && gpio != lookup_gpio_by_name("PWM1") // CHIP PRO
       && !(gpio >= lookup_gpio_by_name("XIO-P0") && gpio <= lookup_gpio_by_name("XIO-P7"))) {
     PyErr_SetString(PyExc_ValueError, "Edge Detection currently available on AP-EINT1, AP-EINT3, and XIO-P0 to XIO-P7 only");
     return NULL;
   }
`

C0deGyver, 2017-06-10:
My project depends on having edge detection on the CHIP Pro GPIO's. Would love to see this implemented!

xtacocorex, 2017-06-10:
This is already implemented.  I left this open as I was expecting @jdinard to respond back with a "hey, you can close this"

I'm actually using edge detection on my CHIP Pro at the moment inside GadgetOS on EINT24.  I implemented the feature because It was missing and I needed it.

jdinard, 2017-06-10:
You can definitely close this. Its implemented in the latest update. Apologies for the late reply.

closed 2017-06-10 by xtacocorex

C0deGyver, 2017-06-10:
I am getting the following error when attempting to use Edge Detection on the GPIO pins of my pro. Any advice? I am using CSID? where ? is the number of the pin I want to use.
ValueError: Edge Detection currently available on AP-EINT1, AP-EINT3, and XIO-P0 to XIO-P7 only

jdinard, 2017-06-10:
Use PWM1, I2S-MCLK, or I2S-DI

C0deGyver, 2017-06-10:
So there is no support planned or current for Edge Detection on the actual GPIO pins (CSID0-CSID7)?

jdinard, 2017-06-10:
There can't be because the chip/chip pro itself doesn't support edge detection on CSID pins

xtacocorex, 2017-06-10:
Yes, the CHIP Pro only supports edge detection on PWM1, I2S-MCLK, and
I2S-DI. The CSI pins do not as the R8/GR8 do not support interrupts on
those pins.

If you want edge detection on the CHIP Pro, you need v0.5.8+, earlier
versions do not have the code.