xtacocorex/CHIP_IO #77: binary wheel pypi package?

pztrick, 2017-08-23:
Hi,
I think it could be useful to upload `armhf` binary package to PyPi to avoid native compilation? I do see the `.deb` package in Releases but this could make it more useful for other non-Debian Linux flavors on GR8 and also allow tagging dependencies in `requirements.txt`.

For my part, I am using Python 3.6 so needed to native compile natively anyway. :) But am compiling on other ARM machine to save GR8 disk space. :)

---

setup.cfg:

```
[bdist_wheel]
universal = 1

[metadata]
license_file = LICENSE
```

```
python setup.py sdist bdist_wheel upload
```

xtacocorex, 2017-08-23:
This is a good idea and is something I haven't gotten around to doing.  I'll try to get to this soon.

closed 2017-09-03 by xtacocorex

xtacocorex, 2017-09-03:
Just implemented this and have unstested wheel files pushed to pypi for Python 2.7 and 3.4. Thanks for the help in this!

commit a0b2ac12 (2017-09-03): Adding the ability to generate wheel packages for uploading to pypi, thanks to help from pztrick. This will close #77