xtacocorex/CHIP_IO #78: gpio_export: something went wrong: No such file or directory

joshuadcampbell, 2017-09-01:
Hey Guys! I am unable to use the GPIO connections on my CHIP. Any suggestions?

`root@chip:~# sudo python3
Python 3.4.2 (default, Oct  8 2014, 14:38:51) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import CHIP_IO.GPIO as GPIO
>>> GPIO.toggle_debug()
 ** debug enabled
>>> GPIO.setup("CSID0", GPIO.OUT)
 ** init_module: setup complete **
 ** gpio_allowed: found match
 ** is_this_chippro: total system ram: 491.0 mb
 ** is_this_chippro: we are a chip
 ** gpio_allowed: pin allowed for chip or both and we're a chip
 ** init_r8_gpio_mem: mapping memory **
 ** map_pio_memory: opening /dev/mem **
 ** map_pio_memory: mapping memory **
 ** map_pio_memory: moving to pio registers **
 ** init_r8_gpio_mem: setup complete **
 ** gpio_export **
gpio_export: something went wrong: No such file or directory
 ** gpio_export: creating data struct **
 ** gpio_set_direction: out **
 ** gpio_set_pud: port 4, pin 4, value 0 **
 ** gpio_get_pud: port 4, pin 4 **
 ** open_value_file **
>>> 

`

xtacocorex, 2017-09-02:
Are you properly cleaning up the GPIO before exiting out of the interpreter? 
What happens if you do: `ls /sys/class/gpio`

jeremyeb, 2019-01-12:
I have the same error using python 2.7.9

`root@chip:/home/chip# ls /sys/class/gpio
export  gpiochip0  gpiochip1013  gpiochip1021  unexport`

closed 2020-06-26 by jshcmpbll

gerardogm, 2020-08-18:
I see the same error being displayed when debug mode is active. In my case, GPIO works even when these messages are being displayed.

I think this is caused by a small bug on function gpio_export on event_gpio.c:

```
    ssize_t s = write(fd, str_gpio, len);  e_no = errno;
    if (DEBUG)
        if (e_no)
            printf("gpio_export: something went wrong: %s\n", strerror(e_no));
```
I'm not an expert on C, but according to what I researched you should only check `errno` when you get an erroneous result. In this case, we should check the value returned by `write` first, and if it's different to the value of `len`, then we should check the error code the function set into `errno` (this check is done further in the `gpio_export` function and it passes in my tests). You should expect the stdlib functions to set `errno` only in case of error. In this case, a previous function call set the `errno` variable to "2" (no such file or directory), so when the code above checked `errno`, the error line was printed. I made a test change to print `errno` before and after the write call and confirmed the `errno` code of "2" was there before the call to `write`, so I think we can ignore these errors. I'll try to submit a patch for this, it's pretty simple.

jshcmpbll, 2020-08-18:
Nice work! I haven't used the GPIO pins on my CHIP for anything and with it being so long I can't remember if this issue is the reason I never fully utilized it. Either way nice contribution!