xtacocorex/CHIP_IO #83: Segmentation fault

jedioncrk, 2018-04-14:
I'm running fairly simple code and I am getting seg. fault with sending values out.

`  GPIO.setup("XIO-P1", GPIO.OUT)
 ** gpio_set_direction: out **
 ** open_value_file **
 ** gpio_allowed: found match
 ** is_this_chippro: total system ram: 491.0 mb
 ** is_this_chippro: we are a chip
 ** gpio_allowed: pin allowed for chip or both and we're a chip
 ** gpio_allowed: found match
 ** is_this_chippro: total system ram: 491.0 mb
 ** is_this_chippro: we are a chip
 ** gpio_allowed: pin allowed for chip or both and we're a chip
Segmentation fault
`

`
 # send out GPIO to open or close garage
                GPIO.setup("XIO-P1", GPIO.OUT)
                time.sleep(1)
                GPIO.output("XIO-P1", GPIO.LOW)
`

Here is my CHIP kernel version:
chip@chip:~/chipflaskgarage-master$ uname -a
Linux chip 4.4.13-ntc-mlc #1 SMP Tue Dec 6 21:38:00 UTC 2016 armv7l GNU/Linux

ii  python-chip-io                             0.7.1-1                                                   armhf        A module to control CHIP IO channels

rysi3k, 2018-12-10:
I have same problem.
When I try to set port value as non-root user is fails with seg fault. On root works great.

xtacocorex, 2018-12-29:
This library has to be run as root.